renode/renode #455: Verilator Integration: Fix LimitTimer Reset in VerilatedPeripheral

econnellmc, 2023-04-26:
### Description

Although the Enabled flag of the LimitTimer is set to 'true' in the constructor of VerilatedPeripheral.cs, the default value is 'false'.  This means that every time a Reset is invoked, then the LimitTimer stops and there is no way to restart it.

### Expected behaviour

The LimitBuffer/LimitTimer mechanism is supposed to update a peripheral periodically, so that it's virtual time stays in line with the main Renode instance.  When using a Verilated peripheral, it is possible to break this synchonisation channel by issuing a reset to the peripheral.  This happens as part of connecting a gdb instance with 'init' from Microchip SoftConsole.

### How to reproduce?

https://github.com/econnellmc/renode-issue-reproduction-template

Apologies for not setting up the full Robot framework, but I hope this is sufficient to show the problem.

I have made this example project based on the Microchip Icicle Kit mpfs-timer example:
https://github.com/polarfire-soc/polarfire-soc-bare-metal-examples/tree/main/driver-examples/mss/mss-timer/mpfs-timer-example

mpfs-timer-example.elf is the built artifact from this project when targeting LIM memory
Vtop is created by running the mpfs-timer Verilog source through Verilator (sim_main and Cmake files also provided)
The scripts and Platform files are from the Renode source, but I have added changes for the Verilated mstimer and some commands for initialising the analysers instead of launching GDB

Steps to reproduce in command.txt:
```
Run Renode, the > commands are for the Monitor

> start @test.resc 

Type '1' in the mmuart1 window.
mmuart2 window starts to report timed interrupts.

> sysbus.mstimer Reset

mmuart2 stops.
Typing a number in mmuart1 restarts or switches the test, but no further interrupts occur.
```

### Environment

Linux   5.15.90.1-microsoft-standard-WSL2
Renode, version 1.13.3.19119 (a72a1fa1-202302201037)

### Do you plan to address this issue and file a PR?

Yes, I have a fix for this already - I need the issue ID from this before I can file a PR.
The fix is very simple, I just added a timer.Enabled=true in the VerilatedPeripheral Reset function.  I have tested it with this example project already.

econnellmc, 2023-04-26:
Additional:

In polarfire-soc.repl, I have an absolute path to the location of Vtop.  It will need to be update to replicate the issue locally.

Line 180: 
    simulationFilePathLinux: "/mnt/c/git/Timer_bug_report/Vtop"

mszprejda, 2023-05-10:
Change added in commit [caf2832](https://github.com/antmicro/renode/commit/caf2832f8bd86de7ced05c6d0450b24a03527063).