renode/renode #457: Is there a way to include a multi-file plugin?

e-shreve-ti, 2023-05-03:
I'm looking at developing a plugin, complex enough that I would prefer to do so with multiple source files. If file A depends on file B, I've tried running "include @/path/fileB.cs" and then running "include @/path/fileA.cs", but for the latter I get an error indicating the type (from fileB) does not exist.

Just wanted to quickly check if multi-file plugins are supported for dynamic inclusion or not. Thanks.

e-shreve-ti, 2023-05-03:
I found that simply placing the compiled plugin assembly (dll) in the same location as the renode executable, allows the monitor to see the plugin and it can be enabled from there.

closed 2023-05-03 by e-shreve-ti

PiotrZierhoffer, 2023-05-05:
That's correct. In general, you don't even need a plugin infrastructure. You can rely on AutoLoadCommand for Monitor commands and type extensions (look for `this Emulation emulation` in the codebase)