renode/renode #463: Simulation stops when one machine is paused

ydnatag, 2023-05-16:
### Description

Having two machines, I want to pause one to emulate a power shortage  but all the simulation stops.

```
(A) emulation GetTimeSourceInfo 
Elapsed Virtual Time: 00:00:38.312000
...
(A) emulation GetTimeSourceInfo 
Elapsed Virtual Time: 00:00:39.065400
...
(A) machine Pause
3;1m(A) 22:35:21.6368 [INFO] A: Machine paused.
(A) emulation GetTimeSourceInfo 
Elapsed Virtual Time: 00:00:45.769385
...
(A) emulation GetTimeSourceInfo 
Elapsed Virtual Time: 00:00:45.769385
...
(A) mach set "B"
(B) machine IsPaused 
False
(B) emulation GetTimeSourceInfo 
Elapsed Virtual Time: 00:00:45.769385
...
(B) emulation IsStarted 
True
```

Is there any way to pause a machine without stopping all the emulation?

PiotrZierhoffer, 2023-05-17:
Hi @ydnatag 

Indeed, pausing the machine will stop the flow of time.

I think what you're looking for is `cpu IsHalted true` on machine `A` - this will not stop the flow of time but will disable the CPU  from executing any code.

ydnatag, 2023-05-18:
I'm looking something that stops also some peripherals in the platform too. Is there any way to continue the simulation with one machine paused? Can you tell me where the emulation's stop is done?

Thanks!

ydnatag, 2023-05-31:
I tried to use "cpu IsHalted true/false". That works for stopping the execution and continue again from the same place. Now I need to reset the CPU to start it again, I run `cpu Reset`,  that stops but it never runs again:

I tried several things to start it without success
```
cpu Reset # stops the cpu but it doesn't start again
cpu Start 
cpu IsHalted # false
cpu Resume
runMacro $reset 
```

Am I doing something wrong? How can I reset the processor?