renode/renode #464: Starting point to write USB controller model on SAMD21?

jkorinth, 2023-05-21:
I would like to simulate USB devices on the ATSAMD21, up to the point that I can hook the model up to USB/IP and use it from the host. In particular, I'm interested in simulating USB HID devices, but also mass storage.

I would appreciate any tips or pointers in the right direction:

1. Is this feasible at all? Can the simulation work, even if it is very slow?

2. Are there any good starting points or precursors in that direction in the Renode code base where I could start from?

Thanks a lot!

mithro, 2023-05-22:
I don't work at Antmicro or on Renode but it might be worth checking out the Fomu bootloader (Foboot) example -- https://renode.readthedocs.io/en/latest/tutorials/usbip.html#real-life-scenario-foboot ?

PiotrZierhoffer, 2023-05-23:
@jkorinth I agree with @mithro  - this is a good scenario to follow.

Just to be on the same page - do you want to use ATSAMD21 as a USB device that you can interact with from your host, or connect devices to be used on your board (aka "board is the USB host")? At this moment the first scenario is supported by the [USBIP server infrastructure](https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Extensions/Utilities/USBIP/USBIPServer.cs)

jkorinth, 2023-05-23:
@mithro @PiotrZierhoffer Thanks for the suggestion, I took a look at the fomuboot. As far as I understand that's a board with a microcontroller attached to a small FPGA which runs the actual USB controller as a soft-core/RTL design, right? What I'd like to do is simulate the USB in the SAMD21 itself, so I can use renode to test the same firmware binary that can run on the actual device.

Goal would be to have the device act as a USB HID device, like a keyboard, or a mouse, and attach virtually to the host for testing via USB/IP. So it would attach from renode via USB/IP to the USB host controller.

My current understanding is, that I'd need to implement a renode model for the USB peripheral in the SAMD21 (0x41005000):

https://zephyr-dashboard.renode.io/renodepedia/socs/atsamd21e18a/?view=peripherals

And most likely for the DMAC too (0x41004800), but I guess there are already models for other DMA controllers that I could use as a starting point?

Do you think it's feasible to implement the USB peripheral in C# so that the simulation would actually enumerate as a USB device?

PiotrZierhoffer, 2023-05-26:
@jkorinth That sounds about right. You seem to want to replicate the FOMU tutorial, just on another board, and it makes perfect sense to do it.

You can look at USB implementations here: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/USB/

FOMU uses the ValentyUSB.cs model.

There is also a ton of DMA devices to look at: https://github.com/renode/renode-infrastructure/tree/master/src/Emulator/Peripherals/Peripherals/DMA

If you'd rather not implement it yourself - we offer commercial services supporting Renode uses, feel free to reach out via contact@antmicro.com