renode/renode #466: Is there a way to hide extra info logs on --disable-xwt static analyzers output?

MSMazaya, 2023-05-24:
When I run the command 

```
renode --disable-xwt -e "s resc_file.resc"
```

I get the output as follows

```
...
11:07:24.8140 [INFO] uart: [host: 1.45s (+0.11ms)|virt:      0.18s (+0s)] Hello World
...
```

Is there any way/options that I can supply to remove the extra info so that it could output only the `Hello World`?

I'm working to use Renode for testing automation, I don't use renode-test because I only need the result of the uart in my stdout.

WojMel, 2023-05-24:
Hi @MSMazaya,

With logging UART you can only limit the prefix with:
```
showAnalyzer "uartAnalyzer" uart Antmicro.Renode.Analyzers.LoggingUartAnalyzer
uartAnalyzer TimestampFormat None
```
but you'll still have a `HH:mm:ss.ssss [INFO] uart: [output] ` prepended to the UART's output.

As an alternative you can use pty or socket terminal as described in [UART integration](https://renode.readthedocs.io/en/latest/host-integration/uart.html) or save output to a file with:
```
uart CreateFileBackend @path_to_file
```

MSMazaya, 2023-05-24:
Hi @WojMel, thank you for the answer! I guess I will be using the socket terminal for the solution, but I am still curious about one thing. I saw that sometimes when I have duplicate outputs of uart (using --disable-xwt) renode adds numerical suffixes for my output such as:

```
...
11:07:24.8140 [INFO] uart: [host: 1.45s (+0.11ms)|virt:      0.18s (+0s)] Hello World (2)
...
```

Can I also remove that behaviour?

MSMazaya, 2023-05-24:
That answers my question👍 thanks @WojMel!

closed 2023-05-24 by MSMazaya