renode/renode #468: Could not create TUN/TAP interface, running in dummy mode.

ekawahyu, 2023-05-30:
### Description

Creating tap device generates the following error while trying to emulate STM32F4Discovery:
```
Could not set TUNSETIFF, error: 2
Could not create TUN/TAP interface, running in dummy mode.
```

### Expected behaviour

The following command should create a `tap0` device, but none created:
```
emulation CreateTap "tap0" "tap"
```

### How to reproduce?

```
mach create
machine LoadPlatformDescription @platforms/boards/stm32f4_discovery-kit.repl
emulation CreateSwitch "switch1"
emulation CreateTap "tap0" "tap"
```

### Environment

ZorinOS 16.2 (Ubuntu 20.04 LTS based)
Renode, version 1.13.3.19119 (a72a1fa1-202302201037)

### Additional information

Complete logs:
```
21:28:15.3350 [INFO] Loaded monitor commands from: /opt/renode/scripts/monitor.py
21:28:28.0474 [INFO] System bus created.
21:29:12.7534 [INFO] Reading cache
21:29:12.9777 [INFO] sysbus: Loaded SVD: /tmp/renode-22634/eeaa9430-89d3-483d-a033-95b94c146457.tmp. Name: STM32F40x. Description: STM32F40x.
Could not set TUNSETIFF, error: 2
21:29:52.8331 [INFO] Copying...
21:29:56.5038 [WARNING] Could not create TUN/TAP interface, running in dummy mode.
```

Creating tap device manually with the following command over terminal (before starting Renode) seems to work just fine:
```
sudo ip tuntap add mode tap tap0
```

Complete log when tap0 device was manually created:
```
21:39:47.4001 [INFO] Loaded monitor commands from: /opt/renode/scripts/monitor.py
21:40:11.9988 [INFO] System bus created.
21:40:15.7674 [INFO] Reading cache
21:40:15.9922 [INFO] sysbus: Loaded SVD: /tmp/renode-22904/7ab4ff03-79dc-4057-bcc2-358319e0f3d2.tmp. Name: STM32F40x. Description: STM32F40x.
21:40:22.0910 [INFO] Opened interface tap0.
```

I am not sure if this helps, but when renode is called within VSCode built-in terminal, it gives different error number:
```
Could not set TUNSETIFF, error: 13
21:18:26.3008 [INFO] Copying...
21:18:31.2699 [WARNING] Could not create TUN/TAP interface, running in dummy mode.
```

### Do you plan to address this issue and file a PR?

No

MateuszKarlic, 2023-05-31:
Hi @ekawahyu I believe that this issue is already fixed on `master` branch. Could you grab a nightly build from here: https://builds.renode.io/ and see if creating TAP works for you now?

ekawahyu, 2023-06-01:
@MateuszKarlic I confirm that the nightly build has no issue with creating TAP device (Recode 1.1.13.3.6117). We can close this issue. Thanks.

closed 2023-06-01 by ekawahyu