renode/renode #469: Killing renode when staticAnalyzer output match defined pattern

MSMazaya, 2023-05-30:
Consider the following is example of my `.resc` script.

```
using sysbus

mach create "K210"

machine LoadPlatformDescription @platforms/cpus/kendryte_k210.repl

path add @/

emulation CreateServerSocketTerminal 3456 'term'
connector Connect sysbus.uart term

showAnalyzer uart

sysbus Tag <0x50440000 0x10000> "SYSCTL"
sysbus Tag <0x50440018 0x4> "pll_lock" 0xFFFFFFFF
sysbus Tag <0x5044000C 0x4> "pll1"
sysbus Tag <0x50440008 0x4> "pll0"
sysbus Tag <0x50440020 0x4> "clk_sel0"
sysbus Tag <0x50440028 0x4> "clk_en_cent"
sysbus Tag <0x5044002c 0x4> "clk_en_peri"

# enable uart tx
uart WriteDoubleWord 0x8 0x1

macro reset
"""
    sysbus LoadELF $bin
    # Is there something like quitWhenAnalyzerMatch "END" that I can use to quit renode when the output of staticAnalyzer met some defined pattern?  
"""
runMacro $reset
```

I am using the above renode script for testing automation and reading its output via the socket terminal. Currently, I need to manually kill the process after getting the output from telnet. Is there any way I can kill the process automatically when an output of the analyzer met some defined pattern? Something like `Wait For Line On Uart` for `renode-test`, Thanks!

MSMazaya, 2023-06-12:
Hi @WojMel

Thank you for the answer, I noticed that the hook does not work when I use `--disable-xwt`. Is there any additional things that I need to do for it to work with `--disable-xwt`?

WojMel, 2023-08-29:
Hi @MSMazaya,

Quitting from hooks when using `--disable-xwt` should be fixed since 95d25ea. Could you verify it works for your use case?

PiotrZierhoffer, 2023-10-19:
Closing. If the issue persists, please open a new one

closed 2023-10-19 by PiotrZierhoffer