renode/renode #471: Crash when trying to map memory high in address space

econnellmc, 2023-06-12:
### Description

When I try to map a block of memory at 0x0001_8000_8000_0000, Renode crashes with an Unhandled Exception.

### Expected behaviour

I was hoping that choosing a 64-bit CPU would let me use memory mappings in these upper ranges.  If it is unsupported, a clearer error message would be helpful.

I've copied the Exception message at the bottom of the issue.

I also tried a few different address locations.  It seems that anything above 0x080000000000 will cause a crash, but I haven't exhaustively investigated the exact tipping point.

### How to reproduce?
[memmap.zip](https://github.com/renode/renode/files/11721539/memmap.zip)

There's a repl and a resc file in the zip that can be used to reproduce the issue.
./bin/Renode.exe memmap.resc

### Environment

Please, provide the following information:

- OS Windows 10
- OS also tried in Ubuntu (WSL)
- Renode (renode-latest from last week), version 1.13.3.33118 (adeb0173-202306051822)

### Do you plan to address this issue and file a PR?

No plans to fix the issue - mostly looking for clarification if my intended usage is supported.

Cheers, Ed

### Error output

$ ./bin/Renode.exe memmap.resc 
11:32:01.0849 [INFO] Loaded monitor commands from: C:\Git\polarfire2-bare-metal-sdk\renode-latest\scripts/monitor.py
11:32:01.6654 [INFO] Including script: C:\Git\polarfire2-bare-metal-sdk\renode-latest\memmap.resc

Unhandled Exception: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at TlibMapRange(RiscV64Wrappers , UInt64 , UInt64 )
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.RegisterMemoryChecked(UInt64 offset, UInt64 size)
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.MapMemory(IMappedSegment segment)
   at Antmicro.Renode.Peripherals.Bus.SystemBus.AddMappings(IEnumerable`1 newMappings, IBusPeripheral owner)
   at Antmicro.Renode.Peripherals.Bus.SystemBus.RegisterInner(IBusPeripheral peripheral, PeripheralAccessMethods methods, BusRangeRegistration registrationPoint, ICPU context)
   at Antmicro.Renode.Peripherals.Bus.SystemBus.Register(IBusPeripheral peripheral, BusRangeRegistration registrationPoint)
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeMethodInfo.UnsafeInvokeInternal(Object obj, Object[] parameters, Object[] arguments)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at Antmicro.Renode.PlatformDescription.CreationDriver.TryRegisterFromEntry(Entry entry)
   at Antmicro.Renode.PlatformDescription.CreationDriver.RegisterFromEntries(IEnumerable`1 sortedEntries)
   at Antmicro.Renode.PlatformDescription.CreationDriver.ProcessInner(String file, String source)
   at CallSite.Target(Closure , CallSite , Object , Object , Object )
   at System.Dynamic.UpdateDelegates.UpdateAndExecuteVoid3[T0,T1,T2](CallSite site, T0 arg0, T1 arg1, T2 arg2)
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberAction(CallSite& callsite, Type binderType, Int32 knownType, LazyBinder binder, InvokeMemberName name, Boolean staticContext, Type context, String[] argNames, Object target, Object[] args)
   at Dynamitey.Internal.Optimization.InvokeHelper.InvokeMemberActionCallSite(Object target, InvokeMemberName name, Object[] args, String[] tArgNames, 
Type tContext, Boolean tStaticContext, CallSite& callSite)
   at Dynamitey.Dynamic.InvokeMemberAction(Object target, String_OR_InvokeMemberName name, Object[] args)
   at Antmicro.Renode.UserInterface.Monitor.InvokeWithContext(InvokeContext context, MethodInfo method, Object[] parameters)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteDeviceAction(Type type, String name, IEnumerable`1 p)
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceAction(Type device, String name, IEnumerable`1 p, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ProcessDeviceActionByName(String name, IEnumerable`1 p, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.TryExecuteScript(String filename, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Commands.IncludeFileCommand.Run(ICommandInteraction writer, PathToken path)
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeMethodInfo.UnsafeInvokeInternal(Object obj, Object[] parameters, Object[] arguments)
   at System.Reflection.RuntimeMethodInfo.Invoke(Object obj, BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at Antmicro.Renode.UserInterface.Monitor.RunCommand(ICommandInteraction writer, Command command, IList`1 parameters)
   at Antmicro.Renode.UserInterface.Monitor.ExecuteCommand(Token[] com, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.ParseTokens(IEnumerable`1 tokensToParse, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.Parse(String cmd, ICommandInteraction writer)
   at Antmicro.Renode.UserInterface.Monitor.HandleCommand(String cmd, ICommandInteraction ci)
   at AntShell.Shell.HandleCommand(String cmd, ICommandInteraction ic)
   at AntShell.CommandLine.HandleControlSequence(ControlSequence seq)
   at AntShell.Terminal.NavigableTerminalEmulator.HandleInput(Object input)
   at AntShell.Terminal.NavigableTerminalEmulator.Run(Boolean stopOnError)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)        
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)

PiotrZierhoffer, 2023-06-19:
We were able to reproduce the issue and we're looking for the underlying cause

mateusz-holenko, 2023-07-25:
Hi @econnellmc, thanks for reporting the issue.

This should be fixed in: 92a3e7932a179b71978ac6dca95cd22351cd7ae2. Could you verify it works for you now?

econnellmc, 2023-08-16:
@mateusz-holenko Apologies for the delay getting back to you on this.  I tested it with a recent version of Renode and it is working as I need it to.  Thanks for the fix!  Ed

closed 2023-08-16 by econnellmc