renode/renode #472: Processing Binary Codes in Renode and Loading onto the Platform

fdsgdshn, 2023-06-15:
Hi,

I've recently started working with Renode. There are a few points that I'm not entirely sure about. I would greatly appreciate your help with these topics. As a newcomer, I have some questions regarding the processing of source code in Renode.

1) As far as I understand, when a code (such as C code) is intended to be loaded into the platform, it first needs to be converted into binary code (.elf, .bin, etc.). For this, I am using a RISC-V compiler (riscv32-unknown-elf-gcc). Then, it can be included in the platform and executed. If a C code depends on other C codes, meaning that these other C codes are also used within the main.c, how does the conversion process to the ELF code work? What approach should I follow for this?

2) Sometimes when I examine ready-made Renode codes, I see that external binary codes are included in addition to the main source code. What is the purpose of this? For example, I came across a .resc code like this:

sysbus LoadBinary @build/software/bios/bios.bin 0
cpu PC 0
sysbus WriteDoubleWord 0xd0228000 0xe3a8
sysbus WriteDoubleWord 0xd0228004 0x96a0f64b
sysbus LoadBinary @build/software/zephyr/firmware.bin 0xd0228008

What is the purpose of the .bin codes included later? Could you briefly explain the operations performed in these codes? 

Thank you very much in advance!

PiotrZierhoffer, 2023-06-19:
Hi @fdsgdshn 

I'll try to answer, but there is nothing really Renode-specific here.

1. First of all, you're not loading C code, you should not think of C code as something runnable by a computer. Code needs to be compiled. The process of compilation is quite a standard procedure: each C file gets translated to an object file, and object files are consolidated into a single elf or bin. You'd need a linker script, and something like a Makefile will be useful to work with multiple input files. Some good sources are [here](https://hackthedeveloper.com/c-program-compilation-process/)

2. I am not sure what are those `WriteDoubleWord` calls for (this is application/platform specific, and I'm not sure what resc script is this), but the important part are `sysbus LoadBinary` and `cpu PC` commands. `LoadBinary` accepts an address as the last address, which means that `bios.bin` is loaded at `0x0` and `firmware.bin` is loaded at `0xd0228008`. These addresses come from the platform architecture and the linker script. 
Two files mean that you have two separate programs - some kind of a bootloader/bios and a Zephyr application. 
`cpu PC` sets the program counter to `0x0`, which means that the simulation will start from `bios.bin`. Its purpose, usually, is to do some preconfiguration of the hardware and then jump to the actual application.

Hope this helps!

fdsgdshn, 2023-06-19:
@PiotrZierhoffer, these pieces of information are highly valuable and extremely helpful. Thank you very much, they have been incredibly beneficial !

closed 2023-06-19 by fdsgdshn