renode/renode #473: machine Reset during WFI hangs the CPU

ydnatag, 2023-06-15:
### Description

I'm trying to reset the whole machine in a simulation using a Leon3. I found that the CPU hangs after the reset and it never recovers from that, ignoring `cpu Reset`, `cpu Start`, etc.

I also tried `machine RequestReset` and `machine RequestResetInSafeState` with same results.

Trying a lot of things, I found that `cpu ShutdownAsNop true` helps to do a `machine Reset`but the performance decreases a lot to be practically unusable.

 ```
machine GetTimeSourceInfo  
Elapsed Virtual Time: 00:00:00.060378
Elapsed Host Time: 00:00:56.173308
Current load: 2006044
Cumulative load: 930.360528669383
State: Idle
Advance immediately: True
Quantum: 00:00:00.000100
```

### Expected behaviour

Procesor should reset without hanging

### How to reproduce?

```
./renode scripts/single-node/leon3_zephyr.resc
(Leon3) start
(Leon3) machine Reset 
<hanged>
```

```
./renode scripts/single-node/leon3_zephyr.resc
(Leon3) cpu ShutdownAsNop true
(Leon3) start
(Leon3) machine Reset 
<reset works but low performance>
```

### Environment

- Renode version: d84de245827d0274152c803d67f4cd85eff96622

PiotrZierhoffer, 2023-06-19:
This is obviously not expected. I see that disabling `ShutdownAsNop` after reset helps with the performance, but this is not what we'd like to have anyway

PiotrZierhoffer, 2023-06-19:
It does look to be SPARC-specific, I'll rename the title