renode/renode #474: STM32G0B1 Emulator Crash - "Trying to execute code outside RAM"

ItamarEliakim-R, 2023-06-19:
### Description
The emulator crashes when running a simple STM32G0B1 code.

### Expected behaviour
Expected to see a blinking led example and UART2 emulation.

### How to reproduce?
Simple STM32G0 code generated using CubeMX, including GPIO, and UART2.
The code crashes and raises the following error - 
cpu: CPU abort [PC=0x0]: Trying to execute code outside RAM or ROM at 0x00000000.

It seems that the execution isn't related to the HAL initialization, as same error occurs on an empty main function.

### Environment
Ubuntu 20.04
Renode, version 1.13.3.19119 (a72a1fa1-202302201037)

### Additional information

### Do you plan to address this issue and file a PR?
Yes

Itamare4, 2023-06-19:
Link - https://github.com/Itamare4/renode-issue-reproduction-template

PiotrZierhoffer, 2023-06-20:
Hi @Itamare4 , thanks for submitting the issue.

I believe you didn't provide your stm32g0.repl - can you please add it to the repo?

ItamarEliakim-R, 2023-06-20:
Hi @PiotrZierhoffer,

Thanks.
The stm32g0.repl is the default one available here - https://github.com/renode/renode/blob/master/platforms/cpus/stm32g0.repl

ItamarEliakim-R, 2023-06-21:
It seems that the codes fails somewhere around here - 
[image]

@PiotrZierhoffer - Any idea?

Thanks

electrodonkey, 2023-06-21:
hi @ItamarEliakim-R ,

did you have a look at 
`HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK`

i suspect that you would need to mock some register value via TAG as described [here](https://renode.readthedocs.io/en/latest/advanced/writing-peripherals.html) in order to get that HAL_OK return value.

Kind regards,
Dan

ItamarEliakim-R, 2023-06-22:
Hi @electrodonkey,

Apologies for being renode newbie :)
Found it, the issue was with the default initialization of the RCC repl on the STM32G0.
If you remove the RCC block from repl, and load the SVD - everything works properly.

Thank you all.
Itamar

closed 2023-06-22 by ItamarEliakim-R