renode/renode #475: Add Verilator support for 16550-based UARTs

econnellmc, 2023-06-20:
### Description

Add a component to the VerilatorIntegrationLibrary that supports UARTs based on the TI 16550 design.  The existing UART handler throws an exception due to a difference in how the registers are used in the 16550 design.  Address 0x00 is used for TX and RX, but it is also used for the Divisor value in the case when the Divisor Latch Bit is set beforehand in the Line Control Reg.  See page 9 in this spec for an example:

http://caro.su/msx/ocm_de1/16550.pdf

### Usage example

I came across this issue when trying to Verilate the UART from the Microchip Icicle Kit reference design.

### Additional information

### Do you plan to address this issue and file a PR?

Yes, I have working code locally, and am preparing an Issue Reproduction Template for a PR.

econnellmc, 2023-06-20:
For reference, this is how the error presents in Renode.  When the code sees a write to 0x00 it assumes it is a byte to be transmitted and waits for activity on the TX pin output.  When this never happens, it times out and throws the following exception:

terminate called after throwing an instance of 'char const*'
10:50:52.3053 [ERROR] mmuart1: Connection error!
Fatal error:
Exception of type 'Antmicro.Renode.Peripherals.CPU.CpuAbortException' was thrown.
  at Antmicro.Renode.Plugins.VerilatorPlugin.Connection.SocketVerilatorConnection.AbortAndLogError (System.String message) [0x00027] in <a97010a20d684eff88167581dfab8f9e>:0 
  at Antmicro.Renode.Plugins.VerilatorPlugin.Connection.SocketVerilatorConnection.ReceiveLoop () [0x0004b] in <a97010a20d684eff88167581dfab8f9e>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00014] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <aeea2e33ea654cfabae63b60ba0e127a>:0 
  at System.Threading.ThreadHelper.ThreadStart () [0x00008] in <aeea2e33ea654cfabae63b60ba0e127a>:0 

10:50:52.3216 [ERROR] mmuart1: Receive error!

econnellmc, 2023-06-20:
I've hit an issue with the robot script.  I've got a local test.repl and test.resc in my issue-reproduction-templare.  I can run renode (renode-portable 13.1) from the command line and include the resc file, and it all works as expected.  However, when I run renode-test and include the resc via the robot file, then I get the following error.

KeywordException: Could not execute script: There was an error executing command 'machine LoadPlatformDescription @test.repl'Could not find file 'test.repl'.

Is there a way to re-include the current directory in the renode PATH when launching via robot?

econnellmc, 2023-06-20:
The error is different on the CI environment than it is locally - some missing libraries for Verilated executables?

+++++ Starting test 'test.Should Run Test Case'
/home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/Vtop_16550: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/Vtop_16550)
/home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/Vtop_16550: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/Vtop_16550)
/home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/Vtop_16550: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/Vtop_16550)
!!!!! Emulation's state saved to "/home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/snapshots/test.Should_Run_Test_Case.fail.save"
!!!!! Log saved to "/home/runner/work/renode-issue-reproduction-template/renode-issue-reproduction-template/logs/test.Should_Run_Test_Case.log"
Error: +++++ Finished test 'test.Should Run Test Case' in 5.47 seconds with status failed

MateuszKarlic, 2023-07-06:
Hi @econnellmc,

We were able to reproduce your issue with GLIBC in local Docker running `Ubuntu 20.04`. It seems like you built your binary against the newer `libc` than available in the CI. The easiest solution would be trying to run your setup on newer image - it worked for us in `Ubuntu 22.04` docker. You can easily change the container version by modyifying line: https://github.com/econnellmc/renode-issue-reproduction-template/blob/main/.github/workflows/test.yml#L8.
Otherwise, you should recompile your binary on the older operating system, with the older `libc`.

For the robot tests, you could try providing relative paths to your scripts. Below are the proposed changes to your `test.resc` file:
```
diff --git a/test.resc b/test.resc
--- a/test.resc
+++ b/test.resc
@@ -8,13 +8,13 @@ logLevel 3
 using sysbus
 mach create

-machine LoadPlatformDescription @../test.repl
+machine LoadPlatformDescription $ORIGIN/../test.repl

-mmuart1 SimulationFilePath @../Vtop_16550
+mmuart1 SimulationFilePath $ORIGIN/../Vtop_16550
 showAnalyzer mmuart1

 macro reset
 """ 
-    sysbus LoadELF @../test_UART.elf
+    sysbus LoadELF $ORIGIN/../test_UART.elf
 """ 
 runMacro $reset
```
By using `$ORIGIN` you are referencing a path relative to the currently executed script. Similarly, you could use `$CWD` in Monitor to get path relative to the working directory (https://renode.readthedocs.io/en/latest/basic/monitor-syntax.html#relative-paths)

PiotrZierhoffer, 2024-08-29:
Hi @econnellmc 

As https://github.com/renode/renode/pull/476 is closed, please take a look if the solution works for you and if we can close this issue

PiotrZierhoffer, 2024-09-04:
As discussed, closing

closed 2024-09-04 by PiotrZierhoffer