renode/renode #477: Zephyr's STM32 new HAL driver unreliable

dcorbeil, 2023-06-20:
### Description

Hi,

I'm new to Renode and I'm trying to run a basic network example of a Nucleo F429zi that can ping and be pinged by the host. I started off with Zephyr's Zperf example and stripped it down to only the bare minimum with a static ipv4 address with an empty main and only use the serial console. It all runs as expected on the actual hardware. The problem comes when I run it in Renode. Initially I can ping the host from my simulated Nucleo F429 but when I ping the simulated board from it works for a few seconds and then get this error in my simulation:

```
*** Booting Zephyr OS build zephyr-v3.3.0 ***
[00:00:00.011,000] <inf> net_config: Initializing network
[00:00:00.011,000] <inf> net_config: Waiting interface 1 (0x20000d00) to be up...
[00:00:00.504,000] <inf> net_config: Interface 1 (0x20000d00) coming up
[00:00:00.504,000] <inf> net_config: IPv4 address: 192.1.2.4
uart:~$ net ping 192.1.2.10
PING 192.1.2.10
28 bytes from 192.1.2.10 to 192.1.2.4: icmp_seq=1 ttl=64 time=0 ms
28 bytes from 192.1.2.10 to 192.1.2.4: icmp_seq=2 ttl=64 time=0 ms
28 bytes from 192.1.2.10 to 192.1.2.4: icmp_seq=3 ttl=64 time=0 ms
[00:00:08.016,000] <err> eth_stm32_hal: HAL_ETH_TransmitIT tx_int_sem take timeout
uart:~$ 
[00:00:20.089,000] <err> eth_stm32_hal: HAL_ETH_TransmitIT tx_int_sem take timeout
[00:00:20.424,000] <err> eth_stm32_hal: HAL_ETH_TransmitIT tx_int_sem take timeout
[00:00:20.757,000] <err> eth_stm32_hal: HAL_ETH_TransmitIT tx_int_sem take timeout
[00:00:21.090,000] <err> eth_stm32_hal: HAL_ETH_TransmitIT tx_int_sem take timeout
[00:00:21.398,000] <err> eth_stm32_hal: HAL_ETH_Transmit: failed!
[00:00:21.742,000] <err> eth_stm32_hal: HAL_ETH_Transmit: failed!
[00:00:21.817,000] <err> eth_stm32_hal: HAL_ETH_Transmit: failed!
[00:00:22.086,000] <err> eth_stm32_hal: HAL_ETH_Transmit: failed!
```

I suspect that the simulated interface doesn't properly process the RX queue and then holds the TX process. I also discovered that when turning off `CONFIG_ETH_STM32_HAL_API_V2` (aka the "new HAL driver"), this issue goes away. 

### Expected behaviour

Expected:
  - Be able to indefinitely have the host and the simulated F429zi ping each other when turning on `CONFIG_ETH_STM32_HAL_API_V2`

Actual behavior:
  - The TX task blocks and ping stops functioning

### How to reproduce?

Here are my config and Renode script:

prj.conf:
```
CONFIG_NETWORKING=y
CONFIG_NET_LOG=y
CONFIG_NET_ARP=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_STATISTICS=y

CONFIG_INIT_STACKS=y
CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_SHELL=y

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.1.2.4"

CONFIG_LOG=y
CONFIG_SHELL_CMDS_RESIZE=n

CONFIG_ETH_STM32_HAL_API_V2=y
CONFIG_ETH_STM32_HW_CHECKSUM=y
```

f429zi_host_ethernet.resc:
```
:name: Run on Nucleo F429zi
:description: This script runs basic ethernet with host

$name?="Nucleo F429zi"
# Change to correct .elf file if necessary.
$bin?=$ORIGIN/../../build/nucleo_f439zi/zephyr/zephyr.elf
$plat?=@platforms/boards/stm32f4_discovery-kit.repl
$tap?="tap0"

using sysbus

emulation CreateSwitch "switch"
emulation CreateTap $tap "tap" true
connector Connect host.tap switch

# Uncomment for launching Wireshark on start
#emulation LogEthernetTraffic

mach create $name
machine LoadPlatformDescription $plat

showAnalyzer usart3

connector Connect sysbus.ethernet switch

macro reset
"""
    sysbus LoadELF $bin
"""

runMacro $reset
```

There is no user code since the only thing needed is access to the uart console. 

### Environment

Please, provide the following information:

- Ubuntu 20.04
- Renode [v1.13.3](https://github.com/renode/renode/tree/v1.13.3)
- Zephy [v3.3.0](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v3.3.0)

### Do you plan to address this issue and file a PR?

I need support on this. I'm new to Renode so I might be missing something.

nvincent-vossloh, 2023-10-03:
Hello, same issue here with 
- Renode, version 1.14.0.29506 (81da46c1-202308081623)
- Ubuntu 20.04.6 LTS
- zephyr v3.4.0

It looks like the board receives the ping request but is unable to send anything and prints the `HAL_ETH_Transmit: failed!` message.

mateusz-holenko, 2023-10-13:
Hi and thanks for reporting the issue.

We've just released some updates to the STM ethernet model (together with a nucleo_f429zi-based Zephyr tests) in https://github.com/renode/renode/commit/8ba95f899629818be9b810d1ac88e2ebb1388848.

Please verify if that resolves problems on your end.

nvincent-vossloh, 2023-11-08:
@mateusz-holenko tested with `renode-1.14.0+20231107git40619bc0.linux-portable.tar.gz` and I manage to ping the board from the host and the host from the board.

nvincent-vossloh, 2023-11-08:
I tried the same tests for a stm32h743 platform and it does not work, here are the steps for reproducing:
```
 west build -b nucleo_h743zi -d vog-zephyr-nodes/build/nucleo_h743zi/zperf zephyr/samples/net/zperf/
```

setup network:
```
sudo ip link add name renode-br0 type bridge
2sudo ip addr add 192.0.2.2/24 brd + dev renode-br0
sudo ip link set renode-br0 up
sudo ip link set tap0 master renode-br0
sudo ip link set tap0 up
```

renode script (adapt path to binary):
```
:name: Run on Nucleo H743
:description: This script runs zperf
$name?="Nucleo H743zi"
# Change to correct .elf file if necessary.
$bin?=@h7_zperf.elf
$plat?=@platforms/cpus/stm32h743.repl

$tap?="tap0"

using sysbus

emulation CreateSwitch "switch"
emulation CreateTap $tap "tap" true
connector Connect host.tap switch

# Uncomment for launching Wireshark on start
#emulation LogEthernetTraffic

mach create $name
machine LoadPlatformDescription $plat

showAnalyzer usart3

connector Connect sysbus.ethernet switch

macro reset
"""
    sysbus LoadELF $bin
"""

runMacro $reset
```
It display this error message: 
```
There was an error executing command 'connector Connect sysbus.ethernet switch'
Parameters did not match the signature
```

I looked at `stm32h743.repl` and if I add the following lines:
```
ethernet: Network.SynopsysEthernetMAC @ sysbus 0x40028000
    -> nvic@61
```

With that, I can start the simulation however my terminal running renode is flooded with: 
```
14:17:14.6384 [WARNING] ethernet: Unhandled read from offset 0x1160. (3)
14:17:14.6384 [WARNING] ethernet: Unhandled read from offset 0xB4.
14:17:14.6385 [WARNING] ethernet: Unhandled read from offset 0xB0. (2)
14:17:14.6386 [WARNING] sysbus: [cpu: 0x8023196] (tag: 'EXTI') ReadDoubleWord from non existing peripheral at 0x580000A8, returning 0x00000000.
```

MateuszKarlic, 2024-02-02:
The Ethernet's model in `stm32h7` is different from the one on `stm32f4` - so it's not possible to reuse this model, as they are not compatible. Zephyr supports them both via STM HAL.

We've recently added [`nucleo_h753zi`](https://github.com/renode/renode/blob/master/platforms/boards/nucleo_h753zi.repl) board REPL to Renode, with Ethernet MAC and PHY. You can refer to a sample networking test here: https://github.com/renode/renode/blob/3c36ec4a5c94edb12fa3862f5ce7b3c38773578b/tests/platforms/nucleo_h753zi.robot for reference. You shouldn't use `stm32h753.repl` directly, as it doesn't have PHY (it's a SoC model, so it lacks a physical Ethernet connector).

While you are targeting `nucleo_h743zi`, this platform should be compatible with `753` in your use case (since `753` has all features of `743` and additional crypto acceleration and security blocks), and you should have no problem using the Ethernet peripheral.

closed 2024-03-21 by PiotrZierhoffer