renode/renode #478: GPIO logic analyzer view

electrodonkey, 2023-06-21:
### Description

Connecting simulator GPIO state to external visualization tool (web page or app)

### Usage example

I imagine seeing the behavior of the firmware not only in the simulator but also visually on a timing diagram/logic analyzer view or chip package view.

### Additional information

Can something like the CreateServerSocketTerminal be used to expose GPIO state?

### Do you plan to address this issue and file a PR?

I would like to work on this but not sure where to start, so any pointers would be helpful.

Spudmn, 2023-07-14:
Have a look at this.
[https://github.com/Spudmn/renode_Logic_Analyzer](url)
This is my first attempt at renode but it seems to work. It creates a Socket server to send the state changes of the pins over TCP/IP.   You will need to write your own GUI using Python or what ever you want.

PiotrZierhoffer, 2023-10-20:
@electrodonkey visualization is definitely in scope, but we have not implemented it yet in a generic way.

You can see one of our previous approaches at the topic here: https://github.com/antmicro/renode-board-visualization/ - but it's quite target specific (although easily expandable).

Having said that, we are currently working on a new Python interface that will make this kind of integration and inspection much easier.
We will definitely write about it on our [blog](https://antmicro.com/blog), please stay tuned.