renode/renode #480: Unresolvable errors gpio & usart stm32f101

Pascal66, 2023-07-07:
Unable to understand thoses errors : 

### Description

```
09:46:08.4949 [INFO] System bus created.
09:46:08.8287 [INFO] sysbus: Loaded SVD: D:\Program Files\Renode\firmware\stm32f101_patched.svd. Name: STM32F101. Description: STM32F101.
09:46:08.9602 [WARNING] gpioPortA: Unhandled write to offset 0x0. Unhandled bits: [14] when writing value 0x44441444. Tags: CNF3 (0x1).
09:46:08.9602 [WARNING] gpioPortC: Unhandled write to offset 0x4. Unhandled bits: [6] when writing value 0x44444414. Tags: CNF9 (0x1).
09:46:08.9602 [WARNING] gpioPortC: Trying to set the state of the input pin #7
09:46:08.9602 [WARNING] gpioPortC: Unhandled write to offset 0x0. Unhandled bits: [30-31] when writing value 0x84444444. Tags: CNF7 (0x3).
09:46:08.9615 [WARNING] gpioPortC: Trying to set the state of the input pin #0
09:46:08.9615 [WARNING] gpioPortC: Unhandled write to offset 0x0. Unhandled bits: [2-3] when writing value 0x44444448. Tags: CNF0 (0x3).
09:46:08.9625 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [6] when writing value 0x44444414. Tags: CNF9 (0x1).
09:46:08.9625 [WARNING] gpioPortA: Unhandled write to offset 0x4. Unhandled bits: [2] when writing value 0x44444441. Tags: CNF8 (0x1).
09:46:08.9625 [WARNING] gpioPortB: Trying to set the state of the input pin #12
09:46:08.9635 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [18-19] when writing value 0x44484454. Tags: CNF12 (0x3).
09:46:08.9635 [WARNING] gpioPortB: Trying to set the state of the input pin #13
09:46:08.9635 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [22-23] when writing value 0x44844454. Tags: CNF13 (0x3).
09:46:08.9635 [WARNING] gpioPortB: Trying to set the state of the input pin #14
09:46:08.9645 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [26-27] when writing value 0x48444454. Tags: CNF14 (0x3).
09:46:08.9645 [WARNING] sysbus: [cpu: 0x801D5C8] WriteDoubleWord to non existing peripheral at 0x42420480, value 0x1.
09:46:09.0075 [WARNING] sysbus: [cpu: 0x8019604] WriteDoubleWord to non existing peripheral at 0x420E0020, value 0x1.
09:46:09.0085 [WARNING] gpioPortB: Unhandled write to offset 0x4. Unhandled bits: [10-11] when writing value 0x44444954. Tags: CNF10 (0x3).
09:46:09.0095 [WARNING] usart3: Unhandled read from offset 0x14. (2)
09:46:09.0120 [WARNING] usart3: Unhandled write to offset 0x14, value 0x0.
09:46:09.0120 [WARNING] usart3: Unhandled read from offset 0x14.
09:46:09.0120 [WARNING] usart3: Unhandled write to offset 0x14, value 0x0.
09:46:09.0130 [WARNING] usart3: Unhandled read from offset 0x14.
09:46:09.0130 [WARNING] usart3: Unhandled write to offset 0x14, value 0x0.
09:46:09.0130 [WARNING] gpioPortA: Unhandled write to offset 0x0. Unhandled bits: [22-23] when writing value 0x44A45444. Tags: CNF5 (0x3).
09:46:09.0130 [WARNING] gpioPortA: Unhandled write to offset 0x0. Unhandled bits: [30-31] when writing value 0xA4645444. Tags: CNF7 (0x3).
09:46:09.0130 [WARNING] gpioPortA: Unhandled write to offset 0x0. Unhandled bits: [26-27] when writing value 0x6A645444. Tags: CNF6 (0x3).
09:46:09.0140 [WARNING] gpioPortA: Unhandled write to offset 0x0. Unhandled bits: [18] when writing value 0x66615444. Tags: CNF4 (0x1).
09:46:09.0140 [WARNING] gpioPortA: Unhandled write to offset 0x0. Unhandled bits: [14] when writing value 0x66651444. Tags: CNF3 (0x1).
09:46:14.1796 [INFO] SPI_SR: [cpu: 0x801FF64] ReadUInt16 from 0x0 (unknown), returned 0x2. (15)
```

### Environment

- OS W11
- Renode version (with commit SHA) Renode, version 1.13.3.20940 (a72a1fa1-202302201136)

### Additional information

Relevant config : 
```
nvic: IRQControllers.NVIC @ sysbus 0xE000E000
    priorityMask: 0xF0
    systickFrequency: 72000000
    IRQ -> cpu@0
cpu: CPU.CortexM @ sysbus
    cpuType: "cortex-m3"
    nvic: nvic
gpioPortA: GPIOPort.STM32F1GPIOPort @ sysbus <0x40010800, +0x400> 
    [0-15] -> exti@[0-15]
usart3: UART.STM32_UART @ sysbus <0x40004800, +0x100> 
    -> nvic@39
```

### Do you plan to address this issue and file a PR?

Need support.

PiotrZierhoffer, 2023-07-25:
Hi @Pascal66 ,

In general, these warnings point out that some features in our models are not implemented. This is quite a common situation, as we typically implement models to the extent we're able to test them, and different software requires different features. It's often the case that these warnings indicate some inaccuracy in the model that, in the end, does not affect the software execution in any way, so they can be safely ignored.

These specific warnings indicate we do not have certain registers or bits within these registers supported.
You can hush these warnings with `logLevel 3 gpioPortB` (this will affect the `gpioPortB` peripheral, you can provide other blocks as well).

To trace what actually causes the issue in your application you can see the execution log of your binary. Please see https://renode.readthedocs.io/en/latest/advanced/execution-tracing.html#logging-executed-function-names for details. 

Tracing the execution should point you to the actual source of your issue (and will add context to these unhandled accesses). With this you should be able to figure out if our models miss some features that are actually required by your software.

Pascal66, 2023-07-28:
Thank you, if I understand correctly I can Ignore those warning, or extend the C# class to add the missed features.
In both case this doesnt affect th execution.

closed 2023-07-28 by Pascal66