renode/renode #481: STM32F4 CRC peripheral

tniessen, 2023-07-07:
### Description

I am trying to emulate an STM32F407VGT6 MCU using Renode for the first time. It appears to be working quite well, with the exception of the CRC peripheral. I am using `platforms/boards/stm32f4_discovery-kit.repl`.

I assume that this is either a configuration problem on my side or a known limitation, however, it unfortunately causes the emulated application to produce incorrect results as far as I can tell.

### Additional information

The CRC peripheral can be used, for example, through [ST's `stm32f4xx_hal_crc`](https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/b0adc88696d491e49086bf995eb24bfd435af284/Src/stm32f4xx_hal_crc.c) or through [libopencm3's `crc` APIs](https://libopencm3.org/docs/latest/stm32f4/html/group__crc__file.html).

When running firmware that uses the CRC peripheral, the terminal shows warnings such as:

```
19:49:22.5815 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5816 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5816 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5817 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5817 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5817 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5818 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5818 [WARNING] sysbus: Write of value 0x910E0BA4 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5819 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5819 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5819 [WARNING] sysbus: Write of value 0x2E8F137D to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5820 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5820 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5821 [WARNING] sysbus: Write of value 0x910E0BA4 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5821 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5822 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5822 [WARNING] sysbus: Write of value 0x2E8F137D to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5823 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5823 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5824 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5824 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5825 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5825 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5826 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5826 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5826 [WARNING] sysbus: Write of value 0x910E0BA4 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5827 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5827 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5828 [WARNING] sysbus: Write of value 0x2E8F137D to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5828 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5828 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5829 [WARNING] sysbus: Write of value 0x910E0BA4 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5829 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5830 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5831 [WARNING] sysbus: Write of value 0x2E8F137D to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5831 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5832 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5832 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5832 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5833 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5833 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5833 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5834 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5835 [WARNING] sysbus: Write of value 0x910E0BA4 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5835 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5835 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5836 [WARNING] sysbus: Write of value 0x2E8F137D to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5837 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5837 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5838 [WARNING] sysbus: Write of value 0x910E0BA4 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5838 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5839 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5839 [WARNING] sysbus: Write of value 0x2E8F137D to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5839 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5840 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5840 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5841 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
19:49:22.5841 [WARNING] sysbus: Write of value 0x1 to an unimplemented register CRC:CR (0x40023008) generated from SVD.
19:49:22.5842 [WARNING] sysbus: Write of value 0xF78E2254 to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5842 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
[... cut ...]
19:49:22.5844 [WARNING] sysbus: Write of value 0x2E8F137D to an unimplemented register CRC:DR (0x40023000) generated from SVD.
19:49:22.5845 [WARNING] sysbus: Read from an unimplemented register CRC:DR (0x40023000), returning a value from SVD: 0xFFFFFFFF.
```

### Do you plan to address this issue and file a PR?

It is entirely possible that I just missed some configuration step, so I would welcome any hints. If adding the CRC peripheral is not possible through configuration alone, I could try to give it a shot if someone provided me with some pointers.

tniessen, 2023-07-24:
@jan-malek I apologize for pinging you directly. I see https://github.com/renode/renode-infrastructure/commit/498784240a0c01d97a13cc28dfc568a06f5dd3bc but I am unsure if it is what I need or how I would set it up. Could you perhaps provide some pointers?

jan-malek, 2023-07-25:
Hi @tniessen

It seems that the crc peripheral is just not present in the platform, I'll address this in Renode. If You need a quick fix You can try to append this line:
```
crc: CRC.STM32F4_CRC @ sysbus 0x40023000
```
To the platform file [here](https://github.com/renode/renode/blob/master/platforms/boards/stm32f4_discovery-kit.repl)

jan-malek, 2023-07-26:
Hi again

In https://github.com/renode/renode/commit/cc1ae58ab986aaac13280c48f2bd2f63d5972c15 we added the missing crc peripheral to all stm32f4 platforms. Could You check if it fixes the issue?

tniessen, 2023-07-26:
@jan-malek Thank you very much for your help! I built renode from source and the warnings about the CRC peripheral in the renode console are indeed gone!

Unfortunately, the simulation still fails, but it's hard to say why: a complex deterministic operation produces an incorrect result. The same code works on a hardware STM32L432KC (a different Arm Cortex-M4 model), and other configurations of the same project work on x86 and x64. Maybe I can get my hands on a hardware STM32F4 to see if the problem is in our code or in renode.

In any case, thank you very much!

PiotrZierhoffer, 2023-07-30:
@tniessen is this something you can share with us?

If you could create a test case via https://github.com/renode/renode-issue-reproduction-template and describe the proper result, then we can try to take a look.

Is this some error in calculations or in behavior?

If you can follow the execution and look for issues, there are many features that can help you analyze your execution, e.g. GDB debugging (https://renode.readthedocs.io/en/latest/debugging/gdb.html) and execution tracing (https://renode.readthedocs.io/en/latest/advanced/execution-tracing.html).

Anyway, I'll close this issue, but feel free to open a new one for your unexpected behavior

closed 2023-07-30 by PiotrZierhoffer

akohlsmith, 2023-08-24:
I've created a bare bones CAN python peripheral that is working for STM32H735. I'm using it with X-CUBE-CRYPTOLIB.

```
# crc32mpeg2 is the "reverse" direction that the peripheral defaults to
def crc32mpeg2(msg):
    crc = regs[0]
    for b in msg:
        crc ^= b << 24
        for _ in range(8):
            crc = (crc << 1) ^ regs[4] if crc & 0x80000000 else crc << 1
    return crc & 0xffffffff

reg = request.offset / 4
if request.isInit:
    regs = [System.UInt32(0)] * 0x100
    regs[0] = System.UInt32(0xffffffff)    # DR
    regs[3] = System.UInt32(0xffffffff)    # initial value after the CRC peripheral is reset
    regs[4] = System.UInt32(0x04c11db7)    # polynomial

elif request.isRead:
    request.value = regs[reg]
    #print("read %08x from %d" % (request.value, reg))

else:
    if reg == 0x2:
        if request.value & 0x1:
            regs[0] = regs[3]

    elif reg == 0x0:
        val_in = request.value
        ba = System.BitConverter.GetBytes(val_in)
        ba_rev = ba[::-1]
        crc = System.UInt32(crc32mpeg2(ba_rev))
        #print("crc: was %08x, in %08x, out %08x" % (regs[0], val_in, crc))
        regs[0] = crc

    else:
        #print("write %08x to %d" % (request.value, reg))
        regs[reg] = request.value
```

This could be expanded to take more from the CR to do the correct bit reversing, inversion, etc., but for my needs I only needed the default (out of reset) operation to work. It was particularly frustrating to write because IronPython errors are not intuitive and the documentation on how to do anything non-trivial in Renode is practically nonexistent.

tniessen, 2023-12-28:
@akohlsmith Thank you! I never was able to get the emulated CRC peripheral to work with the new STM32 cryptographic library (CMOX), but your code helped me implement a stub that [fools CMOX to bypass ST's artificial hardware restrictions](https://gist.github.com/tniessen/cc8d0a86d830f5633098a0701021c8ab).

@jan-malek @PiotrZierhoffer Working with embedded systems has often been frustrating but, even though I never got CRC to work properly, renode has been an incredibly valuable tool. It is refreshing to have an easily extendable simulator. Thank you for your hard work on this!