renode/renode #483: Reduce dependency footprint of Renode packages

pjonsson, 2023-07-09:
### Description

I'm just a user of Renode, and installing Renode in a development container for C/C++ brings in nearly 0.5Gb of additional packages:
```
sudo apt-get install --no-install-recommends ./renode_1.13.3_amd64.deb
...
Need to get 82.9 MB/119 MB of archives.
After this operation, 436 MB of additional disk space will be used.
```

It would be nice if the dependencies could be reduced so installing Renode wouldn't increase the size of the image so much.

Just looking at the dependencies for the package, it might be possible to depend on [mono-runtime](https://packages.ubuntu.com/jammy/mono-runtime) instead of [mono-complete](https://packages.ubuntu.com/jammy/mono-complete), so the mono doc/devel packages in mono-complete could be avoided. The [gtk-sharp2](https://packages.ubuntu.com/jammy/gtk-sharp2) package also depends on doc/devel packages, so perhaps direct dependencies on the required non-dev packages could work.

I am not asking to reduce the dependencies of Renode itself, just for the package to avoid installing things that are not required by Renode.

### Usage example

Would reduce the cost for installing Renode in Dockerfiles.

### Do you plan to address this issue and file a PR?

Do not intend to file a PR.

PiotrZierhoffer, 2023-07-25:
Hi @pjonsson ,

This is a good idea (especially using mono-runtime instead of mono-complete), we'll look into this.

One thing that you can easily do yourself right now is to use our portable package instead. It does not require Mono at all, and takes about 50mb.

The latest portable nightly package is available here: https://builds.renode.io/renode-latest.linux-portable.tar.gz

mateusz-holenko, 2024-04-15:
We've recently reduced the size of Renode packages, hope this resolves the issue.

closed 2024-04-15 by mateusz-holenko