renode/renode #484: Is it possible to interact with NRF52840 with BLE over python ?

vChavezB, 2023-07-17:
### Description

I have seen the tutorial to simulate the interaction between two NRF52840 boards with the central_hr and peripheral_hr example. 

In my use case I want to know if I can develop unit tests to verify that the logic of my BLE peripheral works as expected. I could develop a BLE Central (as in the example with the central_hr) and use the ZTEST framework but I would rather develop the integration tests independent of the Zephyr with e.g., python. 

Is there a way to interact with a simulated NRF52840 with the role of a BLE peripheral over a python interface ?

### Usage example

I connect my simulated NRF52840 BLE peripheral 

```
    mach set "peripheral"
    sysbus LoadELF $peripheral_bin 
```

And the connect to python script (BLE Central) to do integration tests, (i.e., a python module that can interact with the BLE medium of renode).

mateusz-holenko, 2023-07-25:
Hi @vChavezB, 

if you are interested in injecting packets to the radio model you could use a following python snippet executed in Monitor:
```
python "self.Machine['sysbus.radio'].ReceiveFrame(System.Array[System.Byte]([0, 1, 2]))"
```

On the other hand, if you want to intercept a packet sent by the radio you could use https://renode.readthedocs.io/en/latest/basic/using-python.html#packet-interception-hooks.

Please also note that for automated testing Renode comes with the Robot framework integration: https://renode.readthedocs.io/en/latest/introduction/testing.html.

vChavezB, 2023-07-25:
The packet radio otpion seem an interesting option. I was looking for something more high-level. E.g., a virtual hci interface which I could connect to any high-level bluetooth library (e.g.. Bluez)

I found an interesting project which is able to do this

https://github.com/google/bumble

and currently I am connecting the native Zephyr posix binary to an Android emulator over a TCP HCI server.

About the robot framework with renode: Its practical to do tests regarding UART  output from two devices. In my case a BLE Central and a BLE peripheral. However, I am not interested in writing code in Zephyr for a BLE Central to test my Zephyr BLE Peripheral. I would rather do integration tests with the real components I interact with (hence a virtual hci is what I am moving towards):

closed 2023-07-25 by vChavezB