renode/renode #487: STM32_UART Read from DR

IshDeshpa, 2023-07-18:
### Description

STM32_UART will not read input correctly, skipping every other character in any passed in string. Additionally, logs are showing an "unhandled read from offset x14" spamming on UART2 (related to #480) causing the test to hang.

### Expected behaviour

Expected behavior is for UART to read all input characters from the user, but it skips every other character when printing it back out (accessing the DR). If "abcdefg" is input, "aceg" will be the read value. Obviously this is a huge issue for test synchronization between code and the robot test, as main communication is done only through UART.

### How to reproduce?

See https://github.com/IshDeshpa/renode-issue-reproduction/tree/stm-uart.

### Environment

Please, provide the following information:

- OS: Ubuntu 22.04
- Renode version (with commit SHA): Fails on 1.13.3 as well as latest main branch (

### Additional information

For some reason running robot-test hangs for a very long time due to the UART offset x14 issue.

### Do you plan to address this issue and file a PR?

I've tried debugging this myself to no avail, so I will need some support on this. The farthest I've gotten is doing some execution tracing, debugging on the main branch. Here are my brief notes on the UART read issue.

**write to UART:**
On a write to UART from the board code. Here, only one call is made.
- WriteByteUsingDWord
  - Calls ReadDWord and then WriteDWord
  - These trigger the callbacks defined in STM32_UART for UART DR

**read from UART:**
On a read from UART (user input to UART). Here, a call to both Read and Write is made.
- ReadWordUsingDWord
  - Calls ReadDWord
- WriteByteUsingDWord
  - Calls ReadDWord again and then WriteDWord
  - Two calls to ReadDWord is what's causing two things to pop off queue and therefore the skipping of messages

I'm also not sure if it's intended behavior that writing to UART calls WriteByte whereas reading from UART calls ReadWord (instead of ReadByte).

On the UART offset x14 issue, I do not know what the problem is there.

xerpi, 2024-04-13:
Same issue here.

pfre01, 2024-08-27:
I see the same behaviour, every second character is skipped when reading from the uart