renode/renode #492: Methods of external programs controlling renode

Joshua-Shaffer, 2023-07-31:
I am working on a project in which I am attempting to control renode via an external program. My primary use case is to have the external program step through the emulation using the `emulation RunFor dt` command. As far as I can gleam from the documentation, it appears that the best approach to achieve this is to use the headless mode and connect via telnet. 

My primary hang-up, though, is that my program must wait for the machine to finish the emulation over the given timestep before proceeding. I have set up my program to wait for the monitor to return the prompt before proceeding, but it appears to always take at least 1 second even when the machine pauses much more quickly than that (as I can tell from log times in the debug window). 

Is there a way to tell my program that the machine has paused fully in a more timely manner than waiting for the monitor to update a second later? Are there other background processes that would cause renode to be in a non-ready state before the 1 second delay no matter what (say even if I'm emulating 100ms at a time)?

PiotrZierhoffer, 2023-08-01:
Hi @Joshua-Shaffer 

I believe your best bet would be to use https://github.com/antmicro/pyrenode/

This is a Python library that allows you to use a similar syntax we have when testing in Renode.

This library is quite rudimentary, but it should do the trick. We are working on an updated version but there is no clear timeline on when it will be released.

You can see some usage examples in Renodepedia. Go to https://zephyr-dashboard.renode.io/renodepedia/boards/hexiwear_k64/?view=software&demo=Hello_World#colab and click the "Colab" button.

The api you'd use would be as follows:
```
from pyrenode import connect_renode, get_keywords
connect_renode()
get_keywords()
ExecuteScript("yourscript.resc")
CreateTerminalTester("sysbus.uart0", timeout=5) # only if you want to actually test the uart output
ExecuteCommand('emulation RunFor "0.1"')
```

Does this cover your use case?

Joshua-Shaffer, 2023-08-01:
Hi @PiotrZierhoffer

Thanks for the timely reply and pointing me towards this library, I was unaware it existed. I set up a python script to test my use case, but as far as I can tell, the `ExecuteCommand('emulation RunFor "0.1"')` call is still returning after >=1 sec despite the renode logger showing a `Machine paused` message much quicker. 

For example, the following for loop:
`for i in range(0,10):`
`    start = datetime.datetime.now()`
`    ExecuteCommand('emulation RunFor "0.1"')`
`    print(datetime.datetime.now() - start)`

repeatedly prints out durations just slightly longer than 1 sec. Even if I use a more granular time step, it is still taking approximately the same amount of time. Is there a floor on the amount of time that the monitor will take before returning the prompt?

PiotrZierhoffer, 2023-08-02:
Hi @Joshua-Shaffer , no, there is no direct relation like that. We'll try to look into this.

The "0.1" parameter refers to virtual time, not real time. It can be slower depending on the software you're running, but 10x increase of time is rather surprising (unless your code is basically polling on timer value)

I understand that if you drop this setup of switch/tap then it works significantly faster?

Joshua-Shaffer, 2023-08-02:
In my example, I am expecting the virtual time to run faster than real-time. If I run `ExecuteCommand('emulation RunFor "10.0"')`, the wall time is <10 sec.

Digging into the Tap emulator code, I did find a timeout on a read call of 1000ms, https://github.com/renode/renode-infrastructure/blob/25de4ce83e6c2187e7d969c2d321af60068ec531/src/Emulator/Extensions/HostInterfaces/Network/LinuxTapInterface.cs#L227
```                    buffer = LibCWrapper.Read(stream.Handle, MTU, 1000, () => token.IsCancellationRequested);```
I modified this to 1ms, compiled from source, and retested my example. At that point I no longer observed the delay in my example from the previous comment.

PiotrZierhoffer, 2023-08-02:
I see! That's a correct analysis, thank you. I didn't see it because I assumed it's an another thread - which is true, but then we wait for the thread to finish before we enter a paused state at the end of RunFor.

We'll be looking into reworking this. Meanwhile, please remember that keeping a low value will slightly increase the cpu usage in longer execution periods. But I don't think it's relevant for your case.

mateusz-holenko, 2024-07-03:
In https://github.com/renode/renode/commit/a4cbd696ebe1cf5ad72d873f987bf7ce015e8c0d we've introduced performance improvements to `LinuxTapInterface` that should address your scenario.