renode/renode #493: Gaisler UART emulators not functioning correctly

Astro-Ian, 2023-07-31:
### Description

We are trying to create an emulator for a Frontgrade UT32M0R500 microcontroller, which has two UARTs based on the Gaisler IP (this has been confirmed in correspondence with Frontgrade). However, the UART output does not show up on the analyzer and the UART registers are not emulated correctly.

### Expected behaviour

See below for the repl and resc files we're using for the UT32M0R500 emulator startup script and memory map. We are trying to get the UART output to show up using the showAnalyzer commands, but nothing shows up. For comparison, I wrote a script that initializes UART0, prints "hello world\r\n" and then goes into an infinite loop, and ran it on the emulator and a devkit so I could compare the behavior of the UART registers using memory watch. The register values both immediately before and after the print looked like this:

Register		On Devkit	On Renode
DATA		0x00000013	0x00000000
STATUS		0x00000086	0x00000006
CONTROL	0x80000003	0x00000000
SCALER		0x00000144	0x44010000
FIFO DEBUG	0x00000000	0x00000000

Comparing the two scaler registers suggests a reversed byte endianness. Swapping the endianness on the emulated registers, and getting rid of the FIFO debug register which is not used in this program, we have:

Register		On Devkit	On Renode
DATA		0x00000013	0x00000000
STATUS		0x00000086	0x06000000
CONTROL	0x80000003	0x00000000
SCALER		0x00000144	0x00000144

Using the [UT32M0R500 Functional Manual](https://caes.com/sites/default/files/documents/Functional-Manual-UT32M0R500.pdf) to decode the values of these registers, we have the following:

- Devkit DATA register is set to the XOFF character; the emulated one is set to NULL
- The devkit's STATUS register has the transmitter FIFO and shift register empty, and 0 frames in the receiver FIFO; the emulated one has the transmitter FIFO and shift register more than half-full, and 12 frames in the receiver FIFO
- Renode's CONTROL register is completely nonresponsive; in particular, the bits for enabling the transmitter and receiver are both cleared, and the bit for "FIFOs available" is also cleared.

I think the fact that the scaler registers match up strongly indicates that the memory map is correct on the emulator and the endianness differences are correctly accounted for. I strongly suspect the CONTROL register, or the functionality behind it, isn't correctly emulated, causing the software to be unable to enable the transmitter or receiver, which is also causing the FIFO and shift register to fill up as indicated by the status register.

### How to reproduce?

#### ut32m0r500.resc

:name: Frontgrade
:description: This script runs the UT32M0R500 emulation

using sysbus
$name?="Frontgrade"
mach create $name

machine LoadPlatformDescription @ut32m0r500.repl

peripherals

machine SetQuantum "0.0001"
emulation SetGlobalAdvanceImmediately true
path add $ORIGIN

$bin?=@HelloWorld.hex

showAnalyzer sysbus.uart0
showAnalyzer sysbus.uart1

macro reset
"""
    sysbus LoadHEX $bin cpu

    cpu PC 0
"""

runMacro $reset

#### ut32m0r500.repl

bootrom: Memory.MappedMemory @ sysbus 0x00000000
    size: 0x8000

reservedmem1: Memory.MappedMemory @sysbus  0x00008000
    size: 0xFF8000

flash: Memory.MappedMemory @ sysbus 0x01000000
    size: 0x10000

reservedmem2: Memory.MappedMemory @sysbus  0x01010000
    size: 0x1EFF0000

sram: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x18000

reservedmem3: Memory.MappedMemory @sysbus 0x20018000
    size: 0x1FFE8000

// APB peripherals
rtc: Timers.STM32F4_RTC @ sysbus <0x40000000, +0x1000>
    AlarmIRQ -> nvic@4

timer1: Timers.STM32_Timer @ sysbus <0x40001000, +0x1000>
    frequency: 10000000
    initialLimit: 0xFFFF

timer2: Timers.STM32_Timer @ sysbus <0x40002000, +0x1000>
    frequency: 10000000
    initialLimit: 0xFFFFFFFF

uart0: UART.GaislerUART @ sysbus <0x40004000, +0x1000>
    -> nvic@14

uart1: UART.GaislerUART @ sysbus <0x40005000, +0x1000>
    -> nvic@15

spi: SPI.STM32SPI @ sysbus <0x40006000, +0x1000>

i2c1: I2C.STM32F7_I2C @ sysbus <0x40009000, +0x1000>

i2c2: I2C.STM32F7_I2C @ sysbus <0x4000A000, +0x1000>

edacsram:  Memory.MappedMemory @ sysbus 0x4000D000
    size: 0x1000

adc: Analog.STM32_ADC @ sysbus <0x4000F000, +0x1000>
    IRQ->nvic@12

gpioPortA: GPIOPort.STM32_GPIOPort @ sysbus <0x40020000, +0x1000>

gpioPortB: GPIOPort.STM32_GPIOPort @ sysbus <0x40021000, +0x1000>

gpioPortC: GPIOPort.STM32_GPIOPort @ sysbus <0x40022000, +0x1000>

can0: CAN.STMCAN @ sysbus <0x40023000, +0x1000>
    [0] -> nvic@[16]

can1: CAN.STMCAN @ sysbus <0x40024000, +0x1000>
    [0] -> nvic@[17]
    master: can0

systemrom: Memory.MappedMemory @ sysbus 0xF0000000
    size: 0x1000

mtbsfr: Memory.MappedMemory @ sysbus 0xF0200000
    size: 0x1000

mtbsram: Memory.MappedMemory @ sysbus 0xF0210000
    size: 0x10000

nvic: IRQControllers.NVIC @ sysbus 0xE000E000
    priorityMask: 0xF0
    systickFrequency: 72000000
    IRQ -> cpu@0

cpu: CPU.CortexM @ sysbus
    nvic: nvic
    cpuType: "cortex-m0+"

iwdg: Timers.STM32F4_IndependentWatchdog @ sysbus 0x40008000
    frequency: 32000

reservedmem4: Memory.MappedMemory @sysbus 0xF0220000
    size: 0xFDE0000

### Environment

Please, provide the following information:

- Ubuntu 18.04
- Renode version 1.13.1.24171 (5246e523-202306271325)

### Do you plan to address this issue and file a PR?

I could certainly help out with this issue if need be, but I would need help.

Astro-Ian, 2023-08-01:
Hello @PiotrZierhoffer,

1. I've emailed the hex file to that email address.
2. Yes, I've enabled logging and the interactions match what I expect. I can see the software initializing the UART, and then I can see all the characters being printed. I can also see that every time the software reads the status and control registers, they match what I saw on the Renode emulator in my results above.
3. Yes, I think this is the reason for the endianness difference as well. I think this endianness difference is being correctly accounted for, however.
4. The control register not working is probably fine - I don't think any of our software really relies on reading that. I'm more concerned about the status register, since that actually does sometimes get used, and the fact that the UART output doesn't show up on the analyzer.

PiotrZierhoffer, 2023-09-07:
As discussed via email, there's a patch that has to be applied manually to make it work in the little-endian setup:

```
--- a/src/Emulator/Peripherals/Peripherals/UART/GaislerUART.cs
+++ b/src/Emulator/Peripherals/Peripherals/UART/GaislerUART.cs
@@ -15,11 +15,12 @@

 namespace Antmicro.Renode.Peripherals.UART
 {
-    [Endianess(Endianess.BigEndian)]
+    [Endianess(Endianess.LittleEndian)]
```

There is no runtime solution at the moment.

closed 2023-09-07 by PiotrZierhoffer