renode/renode #494: STM32G0 SPI DMA

Itamare4, 2023-08-04:
Thanks for taking the time to submit a feature request for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.rst) first.

### Description

Hi.
I am modifying the Infrastracture code to support an STM32G0 SPI DMA interface as RX/TX.
I wonder - has anyone worked on the case or have a working example (either TX/RX)? Did you succeed in triggering the HAL_SPI_RxCpltCallback/HAL_SPI_TxCpltCallback functions?

The steps I have made are as follow:
1. Initialize the repl file to support DMA functionality.
2. Added DMATransmit to the corresponding DMA stream, similar to DMARecieve.
3. Modifying the STM32G0DMA to support reinitialization of the RX buffer after the empty queue.
There is a bug with the NORMAL DMA mode and how it handles emptying the queue. After the queue is empty, it never gets filled again.

### Usage example

Provide information how do you envision the feature to be used and what usage scenarios could be supported by this feature.

### Additional information

Standard STM32G0 SPI DMA using the following initialization:
``
spi2: SPI.STM32SPI @ sysbus 0x40003800
    IRQ -> nvic@26
    DMARecieve -> dma1@1
    DMATransmit -> dma1@2
``

### Do you plan to address this issue and file a PR?

As soon as the issue is fixed, I'll file a PR.

Itamare4, 2023-08-06:
To elaborate a bit on that, for reading the SPI registers ->
Somehow on a physical MCU their registers are initialized as:
```
SPI Instance Information:
  - SPI Peripheral Address: 0x40003800
  - CR1 Register: 0x00000344
  - CR2 Register: 0x00001708
  - SR Register: 0x00000002
  - DR Register: 0x0000FFFF
```

Where on the Emulator, the same initialization becomes - 
```
SPI Instance Information:
  - SPI Peripheral Address: 0x40003800
  - CR1 Register: 0x00000044
  - CR2 Register: 0x00000002
  - SR Register: 0x00000002
  - DR Register: 0x00000000
```

PiotrZierhoffer, 2023-08-10:
Hi @Itamare4,

Thanks for the detailed description. 

However, to be on the same page, it would be the best if you could share your changes in the models, repl, resc and, if possible, sources for the application you're using.

The best way to do it would be via forking https://github.com/renode/renode-issue-reproduction-template and creating a setup that would fail on CI (where it should pass, if Renode models were correct).

Regarding the registers - you're right. We usually implement our peripherals to some extent, depending on our use case and software we're running. We're happy to see more test cases, as they typically enhance our models - if you can share yours we'll try to investigate the topic further.

Itamare4, 2023-09-10:
Thanks. DMA drivers need more work for the TX/RX application.
I will do my best to PR the issue as soon as I finalize the drivers for STM32.

closed 2023-09-10 by Itamare4