renode/renode #495: Adding a new peripheral in Renode.

fdsgdshn, 2023-08-09:
Hi,

I recently started working with Renode. I want to add a new peripheral for a processor, but I'm having trouble understanding how to do this and couldn't find any resources related to it. When installing Renode, I noticed that it comes with basic files like platforms, scripts, and tools, but I couldn't find any .cs (C#) files within these directories. I created .cs and .csproj files for the new peripheral I want to create. Do I need to copy these files to a specific folder to define them in Renode?

Based on what I found online, I executed the following command: $ORIGIN/MyPlatform.cs. However, I received an error afterward:

"Errors during compilation or loading: There were compilation errors: C:\Program Files\Renode\renode-projects\MyPlatform.cs error CS1928: Type `Antmicro.Renode.Core.Structure.Registers.DoubleWordRegister' does not contain a member `WithReservedBits' and the best extension method overload `Antmicro.Renode .Core.Structure.Registers.PeripheralRegisterExtensions.WithReservedBits(this Antmicro.Renode .Core.Structure.Registers.DoubleWordRegister, int, int, uint?)' has some invalid arguments."

I would be immensely grateful if you could guide me through the steps required to define a new peripheral in Renode.

 I appreciate your help very much.

 Thank you.

PiotrZierhoffer, 2023-08-10:
Hi @fdsgdshn, 

If you're looking for our models, you will find them here: https://github.com/renode/renode-infrastructure/tree/master/src/Emulator/Peripherals/Peripherals

You can take one of these files, change the name and the class name and add your own logic.

Then include it with the `include` command, as I believe you do right now. 

No need to create csproj file at all if you want to load in runtime. 

`include` can point to any path you choose, no requirements there.

I can't comment much on the specific issue your having. It just seems there is some error in your code and the compiler complains. How do you call `WithReservedBits` in your code?

fdsgdshn, 2023-08-10:
Hi @PiotrZierhoffer ,

Thank you for your response. 

Is it enough to edit 'include @.../path to .cs' in order to include my own .cs file that I've written into my Renode project, Even if I build Renode from source? 

Regarding the error I've encountered, the line causing the error is: '.WithReservedBits(7, 1, allowedValue: true)'. What exactly is my mistake here? The error states: 'this Antmicro.Renode.Core.Structure.Registers.DoubleWordRegister, int, int, uint?)' has some invalid arguments.

Thank you!

closed 2023-08-14 by fdsgdshn

jorge-rs, 2023-09-05:
Is there any documentation regarding how to actually add a custom peripheral to the list of available peripherals and not necessarily include it (with include @/path/to/it)?

PiotrZierhoffer, 2023-09-07:
@jorge-rs There is no tutorial, but it's fairly simple.

To compile a peripheral alongside Renode, add it somewhere here: https://github.com/renode/renode-infrastructure/tree/master/src/Emulator/Peripherals/Peripherals

And then list the file somewhere around here: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals.csproj#L159

When you clone Renode, the path to that directory is `src/Infrastructure/src/Emulator/Peripherals/`