renode/renode #498: Uploading Renode Docker Image v1.14.0 to Docker Hub

sondrehaug, 2023-08-16:
### Description

Could you please upload a Docker image with the tag `1.14.0` on Docker Hub(https://hub.docker.com/r/antmicro/renode/tags)? Also, is it possible to update the `latest` tag?

This would make it easy to get this version using `docker pull antmicro/renode:1.14.0`, including the build from https://github.com/renode/renode/tree/v1.14.0

PiotrZierhoffer, 2023-08-18:
Hi @sondrehaug 
Thanks for spotting this.

The latest and 1.14.0 tags are there.

As a side note, if you'd like to stay on the bleeding edge, I very much recommend using the `nightly` tag.

It is not 100% the same image, as it relies on our portable build, but is smaller and has the latest functionality.
We will converge these images eventually.

sondrehaug, 2023-08-22:
Hi @PiotrZierhoffer 
Thank you for updating both `latest` and adding the `1.14.0` tag! I will definitely consider using the `nightly` tag as well. I am closing this issue now.

closed 2023-08-22 by sondrehaug