renode/renode #499: Does the `corext-m7` CPU type fully support FPU and floating point instructions?

bgianfo, 2023-08-18:
### Description
I'm working on trying to get a binary targeting the [`STM32F777IIK6`](https://www.mouser.com/datasheet/2/389/stm32f777bi-1851232.pdf) board working under renode. 
I wrote a custom`.repl` file to expose some of the additional devices based on the existing `STM32F7X` definitions (see below). This board uses a Cortex M7 Core. 

While debugging my binary I realized that it is faulting with `UNDEFINSTR - Execution of undefined instruction` when attempting to execute the following floating point instruction during static initialization. 

```
0x0801de36 <+66>:    vrintm.f64      d6, d6
```

I am wondering if:
- The emulator doesn't support this specific instruction for the `cortex-m7` cpu? 
   * There are instructions that appear to be floating point operations executed immediately before this instruction? 
- Do I need configure something to model the presence of an FPU or enable it?
- Perhaps the `corext-m7` cpu type in renode just lacks full FPU support? 
   * I guess what might be referred to as corext-m7f
   * I see some mention via the [`FPUEnabled`](https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Cores/Arm-M/CortexM.cs#L103) flag but I haven't dug too deep yet. 

Any direction would be helpful, thanks! 

### Additional information

I'm testing with renode 1.14.0.
I'm using my custom [STM32F777IIK6.repl](https://github.com/renode/renode/files/12374469/STM32F777IIK6.repl.txt)

bgianfo, 2023-08-22:
I enabled noisy logging of `renode` and I am able to see the instructoins to enable the FPU exeucting, and I see the log messages about the FPU being successfully enabled.

PiotrZierhoffer, 2023-10-26:
Hi @bgianfo 

Thanks for reporting the issue. 

We have recently identified a few missing FPU instructions and are working on adding them  - `vrintm` included.