renode/renode #501: Fedora packages depend on beesu instead of polkit

lindblandro, 2023-08-24:
### Description

RPM packages generated for Fedora and available from renode.io depend on `beesu` for prompting a password for commands that require root privileges. The issue is that `beesu` is a graphical `su` so it always asks for the root password, which isn't set by default. For example, attempting to run (an example from the docs)
```
(monitor) emulation CreateTap "tap0" "tap"
```

causes `beesu` to prompt for the non-existent root password, and there's no way to cancel the operation as the "cancel" button just causes the prompt to be shown again.

The dependencies for each distro packages are specified in `tools/packaging/make_linux_packages.sh` and based on that other distros do get the more modern policykit treatment.

### Usage example
Using a command that requires escalated privileges, such as
```
(monitor) emulation CreateTap "tap0" "tap"
```
will prompt the current user's password to perform privilege escalation. Canceling the operation or failing to get required privileges cancels the operation without side effects.

### Additional information
```
$ lsb_release -a
LSB Version:    :core-4.1-amd64:core-4.1-noarch
Distributor ID: Fedora
Description:    Fedora release 37 (Thirty Seven)
Release:        37
Codename:       ThirtySeven
```
```
$ renode --version
Renode, version 1.14.0.29506 (81da46c1-202308081623)
```
$ sudo dnf remove beesu                                           
Dependencies resolved.                                                                                   
=========================================================================================================
 Package                      Architecture     Version                     Repository               Size
=========================================================================================================
Removing:                                                                                                
 beesu                        x86_64           2.7-42.fc37                 @fedora                  36 k
Removing dependent packages:                                                                             
 renode                       x86_64           1.14.0-1                    @@commandline           133 M
Removing unused dependencies:                                                                            
 gtk-sharp2                   x86_64           2.12.45-16.fc37             @fedora                 2.8 M
 ibm-data-db2                 x86_64           6.12.0-9.fc37               @fedora                  98 k
 libgdiplus                   x86_64           6.1-3.fc37                  @fedora                 536 k
 libglade2                    x86_64           2.6.4-28.fc37               @fedora                 156 k
 mono-complete                x86_64           6.12.0-9.fc37               @fedora                   0  
 mono-core                    x86_64           6.12.0-9.fc37               @fedora                  56 M
 mono-data                    x86_64           6.12.0-9.fc37               @fedora                  17 M
 mono-data-oracle             x86_64           6.12.0-9.fc37               @fedora                 237 k
 mono-data-sqlite             x86_64           6.12.0-9.fc37               @fedora                 206 k
 mono-devel                   x86_64           6.12.0-9.fc37               @fedora                  23 M
 mono-extras                  x86_64           6.12.0-9.fc37               @fedora                 1.5 M
 mono-locale-extras           x86_64           6.12.0-9.fc37               @fedora                 1.0 M
 mono-mvc                     x86_64           6.12.0-9.fc37               @fedora                 1.6 M
 mono-reactive                x86_64           6.12.0-9.fc37               @fedora                 1.4 M
 mono-wcf                     x86_64           6.12.0-9.fc37               @fedora                 3.0 M
 mono-web                     x86_64           6.12.0-9.fc37               @fedora                 8.5 M
 mono-winforms                x86_64           6.12.0-9.fc37               @fedora                 5.1 M
 mono-winfx                   x86_64           6.12.0-9.fc37               @fedora                 845 k
 monodoc                      x86_64           6.12.0-9.fc37               @fedora                  23 M
 screen                       x86_64           4.9.0-2.fc37                @fedora                 974 k
 usermode-gtk                 x86_64           1.114-5.fc37                @fedora                 247 k
```

### Do you plan to address this issue and file a PR?

No plans.

mateusz-holenko, 2023-12-14:
Hi @lindblandro, thanks for reporting the issue!

As suggested, in https://github.com/renode/renode/commit/0b9d4575fc21d03745f368f3c3d4f588810c5805 we've switched dependencies for Fedora package from `beesu` to `polkit` which should address your case.

lindblandro, 2023-12-17:
Thanks a bunch!