renode/renode #502: CanHub - connect to host CAN network?

akohlsmith, 2023-08-24:
I could have sworn I saw this in the documentation, but after scouring it several times I must have been dreaming it.

I know that I can attach an emulated device to a host ethernet network, but I'm wondering if the same can be done with CAN; I'd like to use Linux or OSX as a CanOpen controller and develop CanOpen peripheral code on an emulated STM32 using the fdcan peripheral.

Is this possible and, if so, is it documented anywhere?

akohlsmith, 2023-08-29:
Looking at an issue in the [zephyr repo](https://github.com/zephyrproject-rtos/zephyr/issues/38698), it sounds like renode can in fact connect the emulated device to the host's CAN network, but I don't see any information nor fragments of a renode config which might allow this. I also don't seem to be able to find better documentation than [this page](https://renode.readthedocs.io/en/latest/networking/machine-to-machine-connections.html) on CanHUB, but it doesn't mention anything about interacting the host bus.

Could one of the renode developers give some insight in how to do this?

yassinedevop, 2024-04-15:
I think this might help :
https://github.com/antmicro/renode/blob/master/tests/peripherals/UT32_CAN.robot
https://github.com/antmicro/renode/blob/master/tests/peripherals/MCAN.robot

yassinedevop, 2024-05-08:
There is a solution to this here #598

PiotrZierhoffer, 2024-05-09:
Hi @yassinedevop 
Correct, `CreateSocketCANBridge` is the solution we have just released.

We are going to publish a chapter on this in our docs very soon.