renode/renode #503: Cortex-M4 Integer Divide-by-Zero doesn't trap

rchtsang, 2023-08-24:
### Description:

Cortex-M3/4 Integer Divide by Zero doesn't seem to get trapped when DIV_0_TRP bit is enabled on Configuration and Control Register.

This may be because I am running on Mac M1 Pro (Arm), but I was able to compile renode and other examples appeared to be running fine, so please let me know if this is likely to be the case. I can try to get access to a x86 machine to see if the issue is still there but I figured I'd file the issue before I forget.

### Expected behaviour

Cortex-M3/4 should trigger fault when `sdiv` or `udiv` instructions detect 0 in the denominator if the DIV_0_TRP bit is enabled on the Configuration and Control Register (CCR) in the System Control Block (See section 4.3.7 of Cortex-M4 Generic User Guide), but Renode doesn't appear to fault. 

I believe this is a case where the logic isn't supported in tlib, but wasn't sure so submitted the issue here.

### How to reproduce?

[Reproduction Repo](https://github.com/rchtsang/renode-issue-div-by-zero-artifacts)
Apologies if I didn't quite set up the repo properly, I am not very familiar with Github Actions/Workflows or the Robot Framework. Most of what I observed was manually investigated.

board: nrf52840.repl

### Environment

- Macbook Pro 2021 (Arm M1 Pro): macOS Ventura 13.5
- Renode, version 1.13.3.35686 (a72a1fa1-202308221949)

### Additional information

For context, I am trying to determine what kind of errors Renode supports emulation for (I want to try building a fuzzer plugin for communication peripherals). So I've built a toy example (source and elf in the reproduction repo) that uses uart to try and trigger a divide-by-zero fault. I am aware that by default, Cortex-M doesn't actually treat an integer divide-by-zero as a fault

After running the `.resc` which creates the machine and loads the binary, I run `start` and enter "e" or "E" into the uart monitor. This should trigger an if condition in the source code that leads to a runtime divide-by-0, which should trap. instead, the device continues to loop and listen to uart.

I printed some of the current state (didn't know how to log this in robot framework) in the monitor, which can be seen below:

```
(machine-0) cpu GetRegistersValues
----------------------
|Name     |Value     |
----------------------
|Control  |0x0       |
|BasePri  |0x0       |
|VecBase  |0x0       |
|CurrentSP|0x0       |
|OtherSP  |0x0       |
|FPCCR    |0xC0000000|
|FPCAR    |0x0       |
|FPDSCR   |0x0       |
|CPACR    |0xF00000  |
|PRIMASK  |0x0       |
|FAULTMASK|0x0       |
|SP       |0x2003FFB8|
|LR       |0x2747    |
|PC       |0x274C    |
|CPSR     |0x21000000|
|R0       |0xB       |
|R1       |0x0       |
|R2       |0x1       |
|R3       |0x0       |
|R4       |0x2003FFE0|
|R5       |0x3E44    |
|R6       |0x0       |
|R7       |0x2003FFB8|
|R8       |0x0       |
|R9       |0x0       |
|R10      |0x20030000|
|R11      |0x0       |
|R12      |0x0       |
|R13      |0x2003FFB8|
|R14      |0x2747    |
|R15      |0x274C    |
----------------------
(machine-0) cpu Step 1
0x0000000000002750
(machine-0) cpu GetRegistersValues
----------------------
|Name     |Value     |
----------------------
|Control  |0x0       |
|BasePri  |0x0       |
|VecBase  |0x0       |
|CurrentSP|0x0       |
|OtherSP  |0x0       |
|FPCCR    |0xC0000000|
|FPCAR    |0x0       |
|FPDSCR   |0x0       |
|CPACR    |0xF00000  |
|PRIMASK  |0x0       |
|FAULTMASK|0x0       |
|SP       |0x2003FFB8|
|LR       |0x2747    |
|PC       |0x2750    |
|CPSR     |0x21000000|
|R0       |0xB       |
|R1       |0x0       |
|R2       |0x1       |
|R3       |0x0       |
|R4       |0x2003FFE0|
|R5       |0x3E44    |
|R6       |0x0       |
|R7       |0x2003FFB8|
|R8       |0x0       |
|R9       |0x0       |
|R10      |0x20030000|
|R11      |0x0       |
|R12      |0x0       |
|R13      |0x2003FFB8|
|R14      |0x2747    |
|R15      |0x2750    |
----------------------
(machine-0) cpu DisassembleBlock 0x274c 4
0x0000274c:   fb92f3f3  sdiv r3, r2, r3
(machine-0) sysbus ReadWord 0xE000ED14
0x0010
```

So as expected, the CCR has a 1 written to bit 4, which means that div_0_trp should be enabled.

Is this one of those scenarios where renode doesn't fully emulate all of the Cortex-M logic?
And if so, would anyone be able to tell me what kinds of bugs/crashes renode supports (or direct me to where i can find that information) ? 

Thanks!

### Do you plan to address this issue and file a PR?

I'd be willing to try but I don't know where I should be looking (I think tlib, but not sure).

PiotrZierhoffer, 2023-10-19:
Hi @rchtsang 

We have implemented this functionality, but due to the fact that one of the main use cases for Renode is prototyping, it would need to be an opt-in feature.

We are now working on making it optional.

zyudneko, 2026-09-08:
Hi @rchtsang,

We have implemented division by zero on Cortex-M, and the `DIV_0_TRP` bit is now respected by default. Also see #992.

closed 2026-09-08 by zyudneko