renode/renode #507: Python function for ElapsedVirtualTime

knaupe, 2023-09-02:
I am trying to use Renodes ElapsedVirtualTime feature in my python script but I can't get it to work.

It works in monitor as expected:

```
Renode, version 1.14.0.29506 (81da46c1-202308081623)
(monitor) i $CWD/platform/stm32f3.resc
Starting emulation...
(stm32f3) machine ElapsedVirtualTime 
[Domain = Elapsed Virtual Time: 00:00:00.837200
Elapsed Host Time: 00:00:37.184407
Current load: 47.77
Cumulative load: 44.415201863354
State: WaitingForReportBack
Advance immediately: False
Quantum: 00:00:00.000100, TimeElapsed = 00:00:00.837200]
```

But not with my python function:

```
(stm32f3) elapsed_virtual_time 
There was an error executing command 'elapsed_virtual_time'
Could not find node 'ElapsedVirtualTime', the longest matching was ''.
(stm32f3)
```

My python code:

```python
#!/usr/bin/env python

import time
from Antmicro import Renode

def mc_elapsed_virtual_time():
    result = self.Machine["ElapsedVirtualTime"]
    elapsed_time = float(result.split(":")[1].strip().split(" ")[0])
    return elapsed_time
```

mateusz-holenko, 2023-09-07:
Hi @knaupe, thanks for asking the question!

The `self.Machine["name"]` notation is used to access peripherals registered in the machine, try `self.Machine.ElapsedVirtualTime.TimeElapsed` instead.

One more note is that `mc_` python functions return status code; if you want to display something in Monitor, use `print`.

PiotrZierhoffer, 2023-10-19:
Closing, feel free to reopen if it doesn't help

closed 2023-10-19 by PiotrZierhoffer