renode/renode #508: Help adding RockPro64 to Renode

sand7000, 2023-09-04:
I would appreciate some help in debugging my attempt to add the rockpro64 to renode. I  attempted to generate a repl file for the board using dtstorepl:

https://github.com/antmicro/dts2repl

I tried both the dts file that comes with seL4 and the one included in the renode u-boot dashboard:

https://u-boot-dashboard.renode.io/rockpro64-rk3399--rk3399-rockpro64-uboot/rockpro64-rk3399--rk3399-rockpro64-uboot.zip

They both result in the same repl file except the dashboard result doesn't add the memory so I used the seL4 result. I tried booting the u-boot elf from the dashboard but I see the following error output on the uart:

```
^A^A^A^A^A^A �^X^X/��&&"Sync"hSr"ynScoynhnroousnoe lrAu:"b sSo0 ync0rA0hrbt0o0o"0r0
 0nhaotnu"s  dAlbeor,rht "e shaa"rn 0dSlnedxry,l0enx  :0e sc0r20r00 h000,rx00 00o20
 000s0r000n0 2000oxfb00ub028
x01
0 0 0:A
b�o���r��"t�"S� yhenlcra:h r0o0n0dn0lo0ues0 r0A0b,o0r0 2te"fs breh4a0 n 0dxllr0e2 0
220000r0,0 0e00s0r00 000x00
0f0b00e0400
"Syenlcrh: r0o00n0o00u00s00 00A00b00o lrr t: "00 00h0a00n0fdfflffefcr8,
 esr 0x02000000
"Synchronous Abort" handler, esr 0x02000000
elr: 00000000002faf00 lr : 00000000002faf00
x0 : 0000000000000000 x1 : 00000000ff1a0000
x2 : 00000000ff1a0014 x3 : 0000000000000020
x4 : 00000000ffffffff x5 : 00000000002fa8e8
x6 : 0000000000000030 x7 : 000000000000000f
x8 : 0000000000000000 x9 : 0000000000000000
```

and in the renode terminal I see:

```
./renode
12:05:38.8640 [INFO] Loaded monitor commands from: /home/esandberg/projects/sel4/renode_1.14.0_portable/scripts/monitor.py
12:05:46.3033 [INFO] Including script: /home/esandberg/projects/sel4/renode_1.14.0_portable/rockpro64.resc
12:05:46.3118 [INFO] System bus created.
12:05:46.8583 [WARNING] Generated interrupt isn't a Private Peripheral Interrupt, interrupt identifier: 131 (SharedPeripheral)
12:05:46.8584 [WARNING] Generated interrupt isn't a Private Peripheral Interrupt, interrupt identifier: 132 (SharedPeripheral)
12:05:47.0739 [INFO] sysbus: Loading segment of 810712 bytes length at 0x200000.
12:05:47.0948 [INFO] cpu0: Setting PC value to 0x200000.
12:05:47.0951 [INFO] cpu1: Setting PC value to 0x200000.
12:05:47.0951 [INFO] cpu2: Setting PC value to 0x200000.
12:05:47.0951 [INFO] cpu3: Setting PC value to 0x200000.
12:05:47.0952 [INFO] cpu4: Setting PC value to 0x200000.
12:05:47.0952 [INFO] cpu5: Setting PC value to 0x200000.
12:05:47.1183 [INFO] RockPro64: Machine started.
12:05:47.1456 [WARNING] cpu4.cpu4_timer: Setting the counter frequency register to 0x16E3600 (24000000Hz), which is different than timer frequency (62500000Hz).
12:05:47.1457 [WARNING] cpu3.cpu3_timer: Setting the counter frequency register to 0x16E3600 (24000000Hz), which is different than timer frequency (62500000Hz).
12:05:47.1457 [WARNING] cpu2.cpu2_timer: Setting the counter frequency register to 0x16E3600 (24000000Hz), which is different than timer frequency (62500000Hz).
12:05:47.1457 [WARNING] cpu0.timer: Setting the counter frequency register to 0x16E3600 (24000000Hz), which is different than timer frequency (62500000Hz).
12:05:47.1457 [WARNING] cpu1.cpu1_timer: Setting the counter frequency register to 0x16E3600 (24000000Hz), which is different than timer frequency (62500000Hz).
12:05:47.1457 [WARNING] cpu5.cpu5_timer: Setting the counter frequency register to 0x16E3600 (24000000Hz), which is different than timer frequency (62500000Hz).
12:05:47.1518 [WARNING] sysbus: [cpu1: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0xC00040.
12:05:47.1519 [WARNING] sysbus: [cpu1: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0x3000100.
12:05:47.1519 [WARNING] sysbus: [cpu1: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E21C, value 0xC000800.
12:05:47.1530 [WARNING] sysbus: [cpu0: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0xC00040.
12:05:47.1531 [WARNING] sysbus: [cpu0: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0x3000100.
12:05:47.1531 [WARNING] sysbus: [cpu3: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0xC00040.
12:05:47.1531 [WARNING] sysbus: [cpu4: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0xC00040.
12:05:47.1534 [WARNING] serial1: Unhandled write to offset 0x10, value 0x3.
12:05:47.1534 [WARNING] serial1: Unhandled write to offset 0x8, value 0x7.
12:05:47.1534 [WARNING] serial1: Unsupported write to LIN mode configuration register at offset 0xC.
12:05:47.1535 [WARNING] sysbus: [cpu3: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0x3000100.
12:05:47.1535 [WARNING] sysbus: [cpu0: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E21C, value 0xC000800.
12:05:47.1535 [WARNING] sysbus: [cpu2: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0xC00040.
12:05:47.1535 [WARNING] sysbus: [cpu3: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E21C, value 0xC000800.
12:05:47.1535 [WARNING] sysbus: [cpu4: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0x3000100.
12:05:47.1538 [WARNING] serial1: Unsupported write to LIN mode configuration register at offset 0xC.
12:05:47.1539 [WARNING] sysbus: [cpu2: 0x202C7C] WriteDoubleWord to non existing peripheral at 0xFF77E028, value 0x3000100.
[... 102 lines cut ...]
    cpuType: "cortex-a75"
    cpuId: 4
    genericInterruptController: gic

cpu4_timer: Timers.ARM_GenericTimer @ cpu4
    frequency: 62500000

cpu5: CPU.ARMv8A @ sysbus
    cpuType: "cortex-a75"
    cpuId: 5
    genericInterruptController: gic

cpu5_timer: Timers.ARM_GenericTimer @ cpu5
    frequency: 62500000

gicIrqCombiner: Miscellaneous.CombinedInput @ none
    numberOfInputs: 2
    -> cpu0@0

serial: UART.NS16550 @ sysbus 0xff180000
    ->gic#0@131

serial1: UART.NS16550 @ sysbus 0xff1a0000
    ->gic#0@132
```

sand7000, 2023-09-05:
I found this:

https://renode.io/news/armv8-a-support-in-renode/

which I think explains why this isn't working yet:

> The addition of ARMv8-A support in Renode opens up a world of possibilities and use cases involving wildly popular CPUs like Cortex-A53 ... Due to the novelty of the support, only basic single-core simulation is available

but the rockpro has multiple cores of both cortex a53 and a75, so I guess for now I just need to wait for an update to Renode.

PiotrZierhoffer, 2023-09-07:
Hi @sand7000 

That's correct, there is no multicore support for ARMv8-A yet. Having said that, our common approach is to have a single core in repl and set `status: "disabled"` in DTS for other cores.

It might just work and give you a good approximation of the target config. I am not certain if u-boot has some compile-time assumptions in that regard or if it relies on dts analysis, but I think it's worth trying.