renode/renode #510: How to trigger NVIC interrupt from pythonperipheral?

akohlsmith, 2023-09-06:
In lieu of trying to make `CANHub.cs` able to connect to a socketcan interface, I'm looking instead at emulating enough of the STM32H7 FDCAN peripheral in Python so I can "talk" to the emulated STM32 that way. The part I'm getting tripped up on so far is how to trigger an NVIC interrupt from the pythonperipheral. I don't seem to see a way to do this.

Is there a way to trigger an NVIC interrupt from Python? Failing that, can I execute a Renode command (`sysbus.nvic` or `sysbus.cpu` perhaps) to trigger an interrupt?

PiotrZierhoffer, 2023-09-07:
Hi @akohlsmith 

Python-based peripherals are not fully fledged peripherals, they do not have options to instantiate interrupt objects and connect them via REPL.

Having said that, you can cheat a little bit and try to access NVIC directly:

```
machine['sysbus.nvic'].OnGPIO(YOUR_IRQ_NUMBER, True)
```

PiotrZierhoffer, 2023-10-19:
Closing - if this doesn't help, feel free to reopen or file a new issue

closed 2023-10-19 by PiotrZierhoffer