renode/renode #513: UART analyzer misses early traffic

KajbaM, 2023-09-15:
### Description
Hello! I am having an issue with use of UART analyzer functionality while trying to simulate zephyr RTOS on nrf52840dk. It appears that UART analyzer misses initial activity and takes some time before it actually starts catching output.

### Expected behavior

Firmware is supposed to print out so called "boot banner" during boot-up. That does happen when trying to run firmware on physical nrf52840dk.

However, UART analyzer directly displays shell prompt, waiting for me to input commands. Analyzer also misses potential early log outputs, but I can get them to show up if I force some code to delay them for a bit before trying to print them.
This makes it impossible to see logging output of early initialization code.

### How to reproduce?

I am using zephyrs `sensor_shell` sample with following Kconfig settings:

```text
CONFIG_SHELL=y
CONFIG_BOOT_BANNER=y
CONFIG_SENSOR=y
CONFIG_SENSOR_SHELL=y
CONFIG_SENSOR_LOG_LEVEL_DBG=y

CONFIG_I2C=y
CONFIG_GPIO=y

CONFIG_LOG=y
CONFIG_LOG_MODE_IMMEDIATE=y
```

The only way I can get UART analyzer to catch early logging is to delay it in firmware (add delay before doing anything).

PiotrZierhoffer, 2023-09-21:
Hi @KajbaM 

Are you able to get both the banner and the shell on hardware?

By looking at Zephyr configs I see there are significant differences when I build this sample with CONFIG_SHELL and CONFIG_SENSOR_SHELL enabled or disabled.

I was able to get the boot banner when I added this additional options:

```
CONFIG_LOG_BACKEND_UART=y
CONFIG_LOG_BACKEND_UART_BUFFER_SIZE=1
CONFIG_LOG_BACKEND_UART_AUTOSTART=y
CONFIG_LOG_BACKEND_UART_OUTPUT_TEXT=y
```

These are the default options when shell is disabled.

I'm not claiming that it should not work without them, but at least it's a way to alleviate the issues for now.

KajbaM, 2023-09-22:
Hi @PiotrZierhoffer 
I have tried your suggestion.
I can confirm that I was able to get both the shell and banner, and some debug prints from sensor initialization before boot baner on hardware.

For some reason, I always get boot banner now (I am reviewing if there was any change in code between creation of issue and now that changed this) how ever, I still don't get all the early logs from before boot banner.

disabling CONFIG_SENSOR_SHELL seems to make stuff worse, I don't get boot banner in case I disable it.

But adding suggested additional options you suggested does bring banner and only the last pre-banner log in case sensor_shell is disabled.

Enabling both sensor_shell and suggested additional options seem to give me the same result.

From testing I managed to get some weird behavior of UART missing outputs even on actual hardware with some configurations, but even in those cases, output from renode was different. 

I am looking further into different scenarios, but there definitely seem to be something about renode grabing uart a bit late?

I am trying to use renode for custom hardware driver development, and that is why early UART loging is pretty much must have for me. I will help myself with combination of suggested additional options and delaying driver initialization for now.

PiotrZierhoffer, 2023-10-19:
Hi @KajbaM .

There is no "late grabing uart" anywhere, really. Renode simply prints out all characters that are sent to it - and this can be seen in the logs with `sysbus LogPeripheralAccess uart1`.

What I believe is happening is that when running on hardware you get a slightly different ordering of events, having Zephyr actually execute in a different way. There are no real software guards in Zephyr that would prevent that from happening - there's a bit of a mess when multiple subsystems access same resources.

Of course we'd rather see the exact same execution flow as on hardware - it needs further investigation on our side. 

You can, however, assume that if the uart is configured and the characters are sent to it, they will get printed out.