renode/renode #516: Semihosting UART support

Lesliehsyu, 2023-09-18:
Dear Renode Team,

I tried to use renode to build up a system and develop that using bare metal method. I tried to use the SemihostingUart as the UART for this system and mapping `printf` function to debugging in the first place, however, after I configured the system and loaded an `.elf` file using 'sysbus LoadELF my.elf'. There is nothing showing up in the monitor or any other places. Since I found that there is a peripheral provided with `SemihotingUart`,  I was wondering if the `SemihostingUart` get supported by renode? If so, where can I find the correct way to open a terminal (or monitor) to observe the output from `SemihostingUart`. I will paste the platform description in the following part.

```
cpu: CPU.CortexM @ sysbus
    cpuType: "cortex-m0+"
    nvic: nvic

uart: UART.SemihostingUart

nvic: IRQControllers.NVIC @ sysbus 0xE000E000
    priorityMask: 0xF0
    systickFrequency: 72000000
    -> cpu@0

Flash: Memory.MappedMemory @ sysbus 0x0
    size: 0x00040000

RAM: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x00020000

```

Looking forward to your reply.

Best regards,
Leslie

Mes-ser, 2023-09-18:
Looking at examples you should define it like that `uart: UART.SemihostingUart @ cpu` in `.repl` file, then in emulation or in `.resc ` put  `showAnalyzer cpu.uart` this will open new window and output anything that goes into "uart"
 
I don't guarantee it will work, but maybe this lead you somewhere

Lesliehsyu, 2023-09-19:
I tried the way @Mes-ser you mentioned, and it shows the `uart` is correctly connected to the system, however, when I run `showAnalyzer cpu.uart`, there is an error shows `Received 'Peripheral not found: cpu.uart' error while initializing analyzer for: cp
u.uart. Are you missing a required plugin?`. Is there any plugin that I needed to install before running this? Thanks!