renode/renode #518: Performance measurement

Nikolai-Hlubek, 2023-09-19:
I have some number crunching code in C++ I would like to optimize. Target platform is an STMF4x. 

Can Renode help? 

1. Interesting for me would be cycle count of a code block. 

However on an emulated STMF4x Discovery board

DWT->CYCCNT always is 0.
Is this not implemented? 

How do I measure cycle count in Renode? 

I tried the collapsed stack view with speedscope which gives some numbers without units or explanation. Are these cycles or something else? 

2. Memory usage

The profiling python script only gives me memory access information. For me
Stack usage over time
Free store usage over time
would be helpful. 

Is it possible to get those values? 

3. Continuous integration with Renode 

The ideal use case would be to include Renode in the continuous integration so that I can compare cycle count and memory usage of a function against a baseline at each build. Is this possible?

Nikolai-Hlubek, 2024-08-15:
Bump. It's been a year. This issue is still relevant for me. 
Does anyone have any idea?

sebastien-riou, 2025-12-05:
It does work on my side:
````
sysbus:
    init:
        Tag <0xE0001000, 0xE0001FFF> "dwt" 
        Tag <0xe000e010 0x10> "systick"

flash: Memory.MappedMemory @ sysbus 0x0
    size: 0x00080000

ram: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x00080000

nvic0: IRQControllers.NVIC @ {
    sysbus new Bus.BusPointRegistration { address: 0xe000e000; cpu: cpu0 }
}
    -> cpu0@0
    systickFrequency: 1000000

cpu0: CPU.CortexM @ sysbus
    cpuType: "cortex-m3"
    nvic: nvic0

dwt: Miscellaneous.DWT @ sysbus 0xE0001000
    frequency: 72000000

uart: UART.LiteX_UART @ sysbus 0x60001800
    -> nvic0@15
````

Do you have it working on real hardware ?
In emulated case, are you sure your code is enabling the DWT ?