renode/renode #519: Dump MappedMemory to file

sullivanmj, 2023-09-22:
### Description

It would be great if we could dump a MappedMemory instance to file for examination. I understand that this can be achieved from within the emulator by writing the memory region to a virtual UART, but this can be slow for large memory regions. There may be other ways of achieving this that I am not aware of.

### Usage example

`sysbus.my_mapped_memory SaveToFile @/home/matt/my_mapped_memory_dump.bin`

### Additional information

N/A

### Do you plan to address this issue and file a PR?

I have a change that seems to be working, that I can submit a PR for.

mateusz-holenko, 2023-10-19:
Hi @sullivanmj, thanks for the question.

In fact that is already possible by using one of python helpers for monitor, namely `dump_file`:
```
(machine-0) dump_file 0 100 @/home/matt/my_mapped_memory_dump.bin
```
This will dump 100 bytes from memory starting at offset 0 to `/home/matt/my_mapped_memory_dump.bin`.

For implementation details see: https://github.com/renode/renode/blob/master/scripts/monitor.py#L70.

PiotrZierhoffer, 2023-10-19:
Please note that the offset is absolute, so if your memory starts at a specific offset you will need to add it to the first parameter of `dump_file`

sullivanmj, 2023-10-25:
Perfect, this is exactly what I needed. Thanks.

closed 2023-10-25 by sullivanmj