renode/renode #524: Zynq-7000 tests failing

shareefj, 2023-10-05:
### Description

I'm just starting to look at your Zynq-7000 support and short of finding any documentation/examples I started trying to run the tests that I could find.  In this case I'm trying to run ``.

What I see is that all tests up to and including `zynq-7000.Should Access SPI Flash Memory Via Additional Cadence xSPI IP` pass and everything after fails.  One of the tests looks like it can't access the UFSHostController, `Error E04: Could not resolve type: 'Storage.UFSHostController'` and the othes seem to fail with `KeywordException: Required state ufs-logged-in not found`.  

What's the current state of these tests?  Should they be working or have you not finished integration yet?

### Expected behaviour

Expecting all committed tests to pass.

### How to reproduce?

```
python3 -m venv venv
. venv/bin/activate
pip install -r tests/requirements.txt
renode-test tests/platforms/zynq-7000.robot
```

### Environment

- Ubuntu 22.04
- Renode, version 1.14.0.29506 (81da46c1-202308081623)

mgielda, 2023-10-11:
HI @shareefj, thanks for reporting the issue.
All Zynq-7000 tests that are shipped with Renode should pass and we can see success the latest CI logs: https://builds.antmicro.com/results/invocations/2af6f5be-4760-4f76-8fc9-d0a5bc25e155?targetFilter=zynq-7000. 

The error you observe looks like a broken Renode build due to desynchronized git submodules (using `renode-infrastructure` before adding the UFS support in https://github.com/renode/renode-infrastructure/commits/6308c0312ec33386712008dcc51deb7d6ef66188). Have you built Renode locally, can you double check if running `git submodule update --recursive` before building helps?

shareefj, 2023-10-11:
Hi @mgielda,

This was a clean clone of renode followed by a `git submodule update --init --recursive`.  The submodules I've got are:

```
git submodule status
 f60a3750ebcd72490b475963a62a779d7a9c730d lib/AntShell (heads/master)
 8f1810e77cf8a3ed9c460a0ee37317cec822d484 lib/BigGustave (v1.0.3-12-g8f1810e)
 bfbfa181fb0768bd8da560882071113a56975274 lib/CxxDemangler (heads/master)
 1a3dbe3f4e76ff624d53d61d9d8275c4a2535416 lib/ELFSharp (heads/renode_base)
 2e14146fcb7397d6f2dac7fac72947398f715a81 lib/FdtSharp (heads/master)
 0c2afc00935c2299ae20e6bda8909531393ea8d4 lib/InpliTftpServer (remotes/origin/renode)
 75d5eaaabd30d28f0f5a1168ad677fd12edeb120 lib/Migrant (v0.4.4-424-g75d5eaa)
 8ddb5d3cd657435ff60b86d30089a0b828a75ff8 lib/Packet.Net (heads/master)
 6b340565473a5375f2b139e26bf04f47849e5fa7 lib/bc-csharp (release-1.9.0-64-g6b340565)
 dc5ba46fe38c5dc62077897a6b40e217d17f0441 lib/cctask (heads/master)
 e355d693da106324388343fdc99c6ebcf7b03b18 lib/options-parser (heads/master)
 65e67ba8de742670fd57aca62f11037d93628404 lib/termsharp (heads/master)
 01e19b676fbab339cc69ef8b8cc46065609e9e3a lib/xwt (heads/master)
 3a3c1fa7d0ad78a98687bd0933d0285ec1de2136 src/Infrastructure (heads/master)
```

and indeed, doing a `git submodule update --recursive` does nothing.  This looks to be the latest commit on master:  75b69f5446cbe17f368aa083bbadbe392b905b5d

shareefj, 2023-10-12:
Looks like it's all fine if I update my clone to the `v1.14.0` tag.  So closing this as user error.  Cheers.

closed 2023-10-12 by shareefj