renode/renode #525: Support STM32G4

qhdwight, 2023-10-09:
### Description

Please add a `.repl` file for STM32 G4 processors.

### Usage example

`machine LoadPlatformDescription @platforms/cpus/stm32g4.repl`

### Additional information

The exact chip we are using is: `STM32G431CBU6`

### Do you plan to address this issue and file a PR?

I have tried to modify the existing [`stm32g0.repl`](https://github.com/renode/renode/blob/master/platforms/cpus/stm32g0.repl) file but to not much luck.

qhdwight, 2023-10-09:
Datasheet can be found here: file:///home/quintin/Downloads/rm0440-stm32g4-series-advanced-armbased-32bit-mcus-stmicroelectronics.pdf

PiotrZierhoffer, 2023-10-18:
Hi @qhdwight,

Thanks for reporting the issue.

While the Renode package does not contain the STM32G4 repl file, we do run this target in our Zephyr Dashboard - https://zephyr-dashboard.renode.io/

If you search for “g4” you will find three targets running plenty of Zephyr samples passing our tests. Clicking the board name will get you here: https://renodepedia.renode.io/boards/nucleo_g431rb/?view=software&demo=hello_world - all the build artifacts, including the REPL platform file, are there to be downloaded.

Please note that the platform description is autogenerated from the device tree file, so it might not cover the whole SoC. It is, however, sufficient to run the demos in the Dashboard.

We’d be happy to learn more about your project - feel free to reach out to [contact@antmicro.com](mailto:contact@antmicro.com) if you can share some details!

qhdwight, 2023-10-21:
Hi @PiotrZierhoffer,

Thank you for the reply! I will check this out.

The application is for a student project team: https://github.com/umrover. We build mock rovers for a global competition that is held over the summer. Part of this is designing a custom brushed motor controller board that uses STM MCU's.

qhdwight, 2023-10-21:
However I noticed that the `.repl` file does not have any `can` devices on it (which we make heavy use of)

. I see CAN mentioned in the `.dts` file but I'm not sure how that plays into everything.

PiotrZierhoffer, 2023-10-26:
Hi @qhdwight 

Indeed, the autogenerated platforms don't have CAN - we don't really need them for the Zephyr samples used in the Zephyr Dashboard.

We do, however, have some CAN support for STM - please scan https://github.com/renode/renode/tree/master/platforms for "STMCAN" to see example usage.

qhdwight, 2023-10-28:
Currently trying out the `.repl` file from the page G431RB you linked. I added several lines for FDCAN based on the datasheet:

```
fdcan1: CAN.STMCAN @ sysbus <0x40006400, +0x400>

fdcan2: CAN.STMCAN @ sysbus <0x40006800, +0x400>

fdcan3: CAN.STMCAN @ sysbus <0x40006C00, +0x400>
```

My `.resc` script is:

```
using sysbus

mach create
machine LoadPlatformDescription @stm32g431.repl
machine StartGdbServer 3333

macro reset
"""
    sysbus LoadELF @Build/bdcmc.elf

"""

runMacro $reset
```

Everything seems to work until `HAL_RCC_ClockConfig` when I receive this error that I can see in the debugger:

[image]

The code can be found here: https://github.com/umrover/mrover-ros/blob/eb560e4c17ffde14cd5f0540b9712e7e723029a5/src/esw/fw/bdcmc/Core/Src/main.c#L90

I will keep debugging. Seems like this is a problem with the emulated RCC.

PiotrZierhoffer, 2023-12-14:
The RCC in this case is definitely quite trivial. 

In this case it only alternates between 0x0 and 0xFFFFFFFF.

You will find some example RCC implementations for STM here: https://github.com/renode/renode-infrastructure/blob/b69fe0ef82f82a82ecce375db1060706ebb97569/src/Emulator/Peripherals/Peripherals/Miscellaneous/ 
(e.g. [STM32F4 RCC](https://github.com/renode/renode-infrastructure/blob/b69fe0ef82f82a82ecce375db1060706ebb97569/src/Emulator/Peripherals/Peripherals/Miscellaneous/STM32F4_RCC.cs)).

Keep in mind that RCC peripherals in ST chips are often quite custom, they can't  be as easily reused as other peripherals.

pmartinf, 2024-02-27:
Hello ...
I have the same issue with 'HAL_RCC_ClockConfig()'. Is there a possibility to get it running ?

GianmarcoCaprile, 2024-11-25:
Hello! Is there any possibility to launch on renode something compiled with STM32CubeIDE? In particular for the board STM32G431KB

GianmarcoCaprile, 2024-11-29:
I'll try to be more precise:

GianmarcoCaprile, 2024-11-29:
I am having some issues trying to launch a simple “Hello world” project which is compiled in STM32CubeIDE, on renode and I would be glad if you could help me. 
At the following link [https://designer.antmicro.com/hardware/devices/weact_stm32g431_core](https://cas5-0-urlprotect.trendmicro.com/wis/clicktime/v1/query?url=https%3a%2f%2fdesigner.antmicro.com%2fhardware%2fdevices%2fweact%5fstm32g431%5fcore&umid=5b75484f-353f-4cd4-91d4-191b08f02b0a&auth=299d746ff6b8e3b1986ad8e7f3a1bda455b6cb4f-da55288a98385bfe7ff3b139dc5fb61964c57cdf)  I found an “hello_world” example project for  STM32G431CoreBoard: this one works on my renode without any problems.
What I am trying to do, instead, is: 
1.	To compile a simple “Hello world” projectin STM32CubeIDE based on STM32G431KBTx micro, to obtain its .elf file. The peripherals enabled within this project are: usart2 and iwdg.
2.	To take the same .repl file found at the link above and to modify it according to my project specification.
3.	To launch them into renode with the following commands: 
•	mach create
•	machine LoadPlatformDescription @scripts/single-node/hello_world-artifacts/hello_world.repl
•	sysbus LoadELF “C:\Programmi\Renode\scripts\single-node\hello_world-artifacts\hello_world.elf”
•	showAnalyzer sysbus.usart2
•	start
But since I see some warning messages in the renode terminal and I can’t see anything on the usart, something goes wrong.
In the attached files you find the  the .repl file I am trying to modify and some screenshot of my STM32CudeIDE setting. 

I attach some screenshoot of my STM32 project

[image]

[image]

[image]

[image]

My .repl file instead is the following: 

sysbus:
    init:
        Tag <0x1f000, 0x1ffff> "storage_partition"
        Tag <0x40000000, 0x40000bff> "timers2 / timers3 / timers4"
        Tag <0x40001000, 0x400017ff> "timers6 / timers7"
        Tag <0x40002c00, 0x40002fff> "wwdg"
        Tag <0x40003800, 0x40003fff> "spi2 / spi3"
        Tag <0x40004800, 0x40004fff> "usart3 / uart4"
        Tag <0x40005400, 0x40005fff> "i2c1 / i2c2 / usb"
        Tag <0x40006400, 0x400067ff> "fdcan1"
        Tag <0x40007800, 0x400083ff> "i2c3 / lptim1 / lpuart1"
        Tag <0x4000a000, 0x4000a74f> "ucpd1 / fdcan1"
        Tag <0x40010400, 0x400107ff> "exti"
        Tag <0x40012c00, 0x40013bff> "timers1 / spi1 / timers8 / usart1"
        Tag <0x40014000, 0x40014bff> "timers15 / timers16 / timers17"
        Tag <0x40020000, 0x40020bff> "dma1 / dma2 / dmamux1"
        Tag <0x40022000, 0x400223ff> "flash"
        Tag <0x48000000, 0x48001fff> "pinctrl"
        Tag <0x50000000, 0x500001ff> "adc1 / adc2"
        Tag <0x50000800, 0x50000bff> "dac1"
        Tag <0x50001000, 0x500013ff> "dac3"
        Tag <0x50060800, 0x50060bff> "rng"
        Tag <0xe000e010, 0xe000e01f> "systick"

flash0: Memory.MappedMemory @ sysbus 0x8000000
    size: 0x20000

sram0: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x8000

clk_lse: Python.PythonPeripheral @ sysbus 0x40007000
    size: 0x4
    initable: true
    filename: "scripts/pydev/rolling-bit.py"

gpioa: GPIOPort.STM32_GPIOPort @ sysbus <0x48000000, +0x400>

gpiob: GPIOPort.STM32_GPIOPort @ sysbus <0x48000400, +0x400>

gpioc: GPIOPort.STM32_GPIOPort @ sysbus <0x48000800, +0x400>

gpiod: GPIOPort.STM32_GPIOPort @ sysbus <0x48000c00, +0x400>

gpioe: GPIOPort.STM32_GPIOPort @ sysbus <0x48001000, +0x400>

gpiof: GPIOPort.STM32_GPIOPort @ sysbus <0x48001400, +0x400>

gpiog: GPIOPort.STM32_GPIOPort @ sysbus <0x48001800, +0x400>

// autogenerated

led0: Miscellaneous.LED @ gpioc 0x6

gpioc:
    6 -> led0@0

nvic0: IRQControllers.NVIC @ {
    sysbus new Bus.BusPointRegistration { address: 0xe000e000; cpu: cpu0 }
}
    -> cpu0@0
    systickFrequency: 144000000

cpu0: CPU.CortexM @ sysbus
    cpuType: "cortex-m4f"
    nvic: nvic0

rcc: Python.PythonPeripheral @ sysbus 0x40021000
    size: 0x400
    initable: true
    filename: "scripts/pydev/flipflop.py"

rtc: Timers.STM32F4_RTC @ sysbus 0x40002800
    AlarmIRQ->nvic0@41

usart2: UART.STM32F7_USART @ sysbus 0x40004400
    frequency: 200000000
    IRQ->nvic0@38

iwdg: Timers.STM32_IndependentWatchdog @ sysbus 0x40003000
    frequency: 32000

// cortex-m overlay

dwt: Miscellaneous.DWT @ sysbus 0xE0001000
    frequency: 72000000

// st,stm32g4 overlay