renode/renode #532: Renode monitor crashes at startup

samiusn, 2023-10-13:
### Description
Version : `Renode, version 1.14.0.34213`
When starting renode, it exceed memory and crashes due to exceedeing memory range.
````
Fatal error:
Array dimensions exceeded supported range.
   at System.Collections.Generic.List`1.set_Capacity(Int32 value)
   at System.Collections.Generic.List`1.EnsureCapacity(Int32 min)
   at System.Collections.Generic.List`1.Add(T item)
   at System.IO.File.InternalReadAllLines(String path, Encoding encoding)
   at AntShell.CommandHistory.Load(String path)
   at AntShell.Shell.Start(Boolean stopOnError)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart(Object obj)
````

### Additional information

I have been working with renode for 1 year but there have been no issues of this sort. My colleagues are also working without any issue as well as of now.

I have tried.
1. install previouse version 1.13.XX (it used to work but now having same issue)
2. Did not tried to built from source.
3. Initially 1.14.0 was working fine for 2 months then suddenly this happened,
4. When I run in ADMIN mode it is working fine.
5. I can run it Docker
6. I cam run test cases using .robot without monitor looks like there is no problem

MateuszKarlic, 2023-10-14:
Hi @samiusn 

It looks like your command history might have grown so large that the Monitor can't handle it anymore.

Can you back up your current history under a different name and try running Renode?
On Windows it should be located under `%appdata%/renode/history` while on Linux under `~/.config/renode/history`

samiusn, 2023-10-19:
Thanks, that works.

closed 2023-10-19 by samiusn