renode/renode #533: USB CDC logging

nobodyguy, 2023-10-15:
Hello, thank you for such a great project!
I have one question - is logging via USB CDC supported? I can't see any mentions.

PiotrZierhoffer, 2023-10-26:
Hi @nobodyguy 

We do have some USB devices supported, but in general, when discussing specific features, we need to do it in the context of a specific SoC. With a USB controller modeled you should be able to configure it as a CDC device and then use it as such. 

Can you say a bit more about your specific use case?

nobodyguy, 2023-10-26:
Hello @PiotrZierhoffer,
thank you for the answer.
My use case:
I have a nRF52840 based board with Zephyr RTOS where logging is redirected to the USB. And I want to write some basic tests for the firmware in the Renode environment without any major configuration changes.

rbborges, 2025-03-06:
Hi! Is this kind o feature supported by any of the existing platforms  in Renode? If so, it may be helpful to know, as someone (maybe myself) can dig into how it has been omplemented, to extend to other cpus / SOC.