renode/renode #536: Adding NXP ADC Peripheral

mjalili96, 2023-10-21:
Hi, How we Can add a  peripheral for NXP boards like I2C or ADC. can we use and stm32 ADC on a NXP board or use its CS file as a template?

PiotrZierhoffer, 2023-10-26:
Hi @mjalili96 

In general, Renode models follow the reference manuals of specific SoCs - you can't really reuse STM ADC on NXP, unless you also use a driver for STM. You can think of our models as "anti-drivers" - they have to match.

You can of course use it as an example of how to write a model, but adjusting the register interface and internal logic will definitely be required.

mjalili96, 2023-10-29:
@PiotrZierhoffer Thanks for your response, what steps should I consider to add NXP ADC Peripheral? what changes in renode i have to make and what files need to be added and developed? actually i want a read map to do it
thanks

PiotrZierhoffer, 2023-12-14:
Hi @mjalili96 

We have documented this process a bit: https://renode.readthedocs.io/en/latest/advanced/writing-peripherals.html

The general idea is to start with listing all registers and their fields as tags (like here: https://github.com/renode/renode-infrastructure/blob/3f1abde88ac5a2dae326b77ab91892f335e78f80//src/Emulator/Peripherals/Peripherals/I2C/MPFS_I2C.cs#L112), run your software, gather information from the Renode log on which fields and registers are actually used by the software, and add implementations of these specific registers.

PiotrZierhoffer, 2024-10-24:
Closing for now - if you have other questions, of course please open further issues

closed 2024-10-24 by PiotrZierhoffer