renode/renode #540: [ARM] Improve the Debug Register Interface of the Coprocessor 14 register interface

vannaka, 2023-10-26:
### Description

Currently, only 3 of the Debug Register Interface registers within the Coprocessor 14 are [supported](https://github.com/antmicro/tlib/blob/705325c09c2ef9ac5dfa3b2f42f5fc222299b8c9/arch/arm/translate.c#L6532). Any access to other CP14 registers results in an undefined instruction abort. The project I'm trying to get running on Renode accesses a CP14 register, which is not supported in Renode, and subsequently takes an undefined instruction cpu exception.

I really just need a single register to read as zero. Specifically, register 34 of CP14, i.e. `DBGDSCR.ext`. The instruction to access this register would be `MRC p14, 0, <Rt>, c0, c2, 2`. See section C6.4.3 of the [ARMv7 Arch Reference Manual](https://developer.arm.com/documentation/ddi0406/cd/).

The better solution would be to determine what combination of registers and their bitfields should be set to indicate the debug interface is out of service and should not be used. Hopefully most projects probe similar registers in this regard.

See section C6 of the ARMv7 Reference Manual for more info on the Debug Register Interface within the CP14 register interface.

### Do you plan to address this issue and file a PR?

Yes, although I noticed [some work](https://github.com/antmicro/tlib/commit/c00adcc189750b93e085ee4ffcde2305317e2d8b) around the coprocessor registers in relation to a `TTable`. I'd like to help with this issue, but it appears my work could be made redundant. Please advise.

PiotrZierhoffer, 2023-10-26:
Hi @vannaka ,

Thanks for reporting the CP14 issues - indeed, we are currently revamping the CP14/15 infrastructure, so I would suggest waiting for our release. This should happen relatively soon.

We'll most likely start with a stub, for now, but using the ttable approach.

MateuszKarlic, 2023-11-10:
Hi @vannaka,

We recently revamped our CP14 infrastructure (1e3e50f6a6da4989965c414651fc07a5e096d76c)  and included `DBGDSCRext` as Read-As-Zero stub.

vannaka, 2023-12-06:
This fixed the CP14 support enough for my needs. Thanks!

closed 2023-12-06 by vannaka