renode/renode #541: [TLIB][ARM] Possible bug with two of the three currently supported CP14 registers

vannaka, 2023-10-26:
## Description

While looking through the codebase for #540, I noticed what appears to be a bug. The comment at [TLIB/arch/arm/translate.c#L6544](https://github.com/antmicro/tlib/blob/705325c09c2ef9ac5dfa3b2f42f5fc222299b8c9/arch/arm/translate.c#L6544) says the code at this location is working with the CP14 registers `DBGDRAR` and `DBGDSAR`.

For reference, the following are the arguments to the `MRC` instruction:
 * `MRC p14, <op1>, <Rt>, <CRn>, <CRm>, <op2>`

### What the comment says the code is supporting
 * DBGDRAR: `MRC p14, 0, <Rt>, c1, c0, 0`
 * DBGDSAR: `MRC p14, 0, <Rt>, c2, c0, 0`

### What the code is supporting
 * `MRC p14, 0, <Rt>, c0, c1, 0` - DBGDSCRint
 * `MRC p14, 0, <Rt>, c0, c2, 0` - Undefined Register

As you can see, it seems the author of this code accidentally swapped the positions of the `CRn` and `CRm` arguments to the `MRC` instruction. The intent was to support reading `DBGDRAR` and `DBGDSAR`, instead `DBGDSCRint`, and an undefined register are read.

### Do you plan to address this issue and file a PR?
Yes.

PiotrZierhoffer, 2023-10-26:
Thanks for reporting this! As mentioned in #540 - we are currently redoing CP14 and we'll release it soon, so please stay tuned. 

Of course we'll take this report into account.

MateuszKarlic, 2023-11-10:
Thank you for reporting this @vannaka, the changes fixing this have just been merged as of 1e3e50f6a6da4989965c414651fc07a5e096d76c.

closed 2023-11-10 by MateuszKarlic