renode/renode #552: Index out of bounds defining multiple flash/sram peripherals

SandyWalsh, 2023-11-16:
### Description

Attempting to define 8 sram and 8 flash (MemoryMapped) peripherals results in unhandled exception. Index out of bounds. 

### Expected behaviour

Should be able to define as many sram perph as needed. 

### How to reproduce?

## REPL
```
(monitor) mach create "sandy"
(sandy) machine LoadPlatformDescription @platforms/boards/my_test.repl
```

```
gpiob: GPIOPort.STM32_GPIOPort @ sysbus <0x58020400, +0x400>

greenled: Miscellaneous.LED @ gpiob 0x0

gpiob:
    0 -> greenled@0

gpioe: GPIOPort.STM32_GPIOPort @ sysbus <0x58021000, +0x400>

yellowled: Miscellaneous.LED @ gpioe 0x1

gpioe:
    1 -> yellowled@0

cpu: CPU.CortexM @ sysbus
    cpuType: "cortex-m7"
    nvic: nvic

nvic: IRQControllers.NVIC @ sysbus 0xe000e000
    -> cpu@0

adc1: Analog.STM32_ADC @ sysbus 0x40022000
    IRQ->nvic@18

gpioa: GPIOPort.STM32_GPIOPort @ sysbus <0x58020000, +0x400>

gpioc: GPIOPort.STM32_GPIOPort @ sysbus <0x58020800, +0x400>

gpiod: GPIOPort.STM32_GPIOPort @ sysbus <0x58020c00, +0x400>

gpiof: GPIOPort.STM32_GPIOPort @ sysbus <0x58021400, +0x400>

gpiog: GPIOPort.STM32_GPIOPort @ sysbus <0x58021800, +0x400>

gpioh: GPIOPort.STM32_GPIOPort @ sysbus <0x58021c00, +0x400>

gpioi: GPIOPort.STM32_GPIOPort @ sysbus <0x58022000, +0x400>

gpioj: GPIOPort.STM32_GPIOPort @ sysbus <0x58022400, +0x400>

gpiok: GPIOPort.STM32_GPIOPort @ sysbus <0x58022800, +0x400>

flash0: Memory.MappedMemory @ sysbus 0x8000000
    size: 0x200000

flash1: Memory.MappedMemory @ sysbus 0x8000000
    size: 0x100000

flash2: Memory.MappedMemory @ sysbus 0x8100000
    size: 0x100000

flash3: Memory.MappedMemory @ sysbus 0x0000000
    size: 0x0
[... 147 lines cut ...]
  at Antmicro.Renode.UserInterface.Monitor.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ci) [0x00000] in <11d556aecc3649618ce70f6971350082>:0 
  at AntShell.Shell.HandleCommand (System.String cmd, AntShell.Commands.ICommandInteraction ic) [0x000bd] in <12b5ceac888f4387a227ffd402a29e15>:0 
  at AntShell.CommandLine.HandleControlSequence (AntShell.Helpers.ControlSequence seq) [0x00aa1] in <12b5ceac888f4387a227ffd402a29e15>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.HandleInput (System.Object input) [0x0002e] in <12b5ceac888f4387a227ffd402a29e15>:0 
  at AntShell.Terminal.NavigableTerminalEmulator.Run (System.Boolean stopOnError) [0x00017] in <12b5ceac888f4387a227ffd402a29e15>:0 
  at AntShell.Shell.Start (System.Boolean stopOnError) [0x00102] in <12b5ceac888f4387a227ffd402a29e15>:0 
  at Antmicro.Renode.UI.CommandLineInterface+<>c__DisplayClass0_1.<Run>b__3 (System.Object x) [0x00000] in <eb0d5c6c2ada4750b5c3a3c345d7e4f8>:0 
  at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x0002c] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x0008d] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x00031] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
  at System.Threading.ThreadHelper.ThreadStart (System.Object obj) [0x00012] in <12b418a7818c4ca0893feeaaf67f1e7f>:0 
```

### Environment

- OS
Ubuntu 20.04.6 LTS
 
- Renode version (with commit SHA)
Renode, version 1.14.0.29506 (81da46c1-202308081623) on Ubuntu 

### Do you plan to address this issue and file a PR?

Still investigatings.

SandyWalsh, 2023-11-17:
I reduced the number of sram entries to 3 and it was fine

```
sram0: Memory.MappedMemory @ sysbus 0x24000000
    size: 0x80000
sram1: Memory.MappedMemory @ sysbus 0x30000000
    size: 0x40000
sram7: Memory.MappedMemory @ sysbus 0x38800000
    size: 0x1000
```

SandyWalsh, 2023-11-17:
Looks like a fencepost error here
https://github.com/renode/renode-infrastructure/blob/c2e1669c54ac8740018b7ce4e8fcfacc867d533d/src/Emulator/Main/Peripherals/Memory/MappedMemory.cs#L533-L561

mateusz-holenko, 2023-11-22:
Hi @SandyWalsh, thanks for reporting the issue.

We believe the main problem here is not the amount of flash/sram peripherals (there should be no hard-limits on the amount of them in Renode), but rather the fact that some of them are defined with size set to 0.
Of course this should not crash Renode and be handled in a more user-friendly way instead - the fix for this will be released soon.

For the record: we also see you have many peripherals defined at the same offset on the system bus, so the presented platform file needs further updates before it can be correctly loaded into Renode.

mateusz-holenko, 2023-11-23:
With https://github.com/renode/renode/commit/44e91265992299da75cacde8d560dddb44168034 you should not observe crashes of Renode anymore.