renode/renode #555: Mac : Unable to run renode with stm32f4_discovery.resc

princehektor, 2023-11-23:
### Description

I am trying to use renode for the first time . Using an M1 Mac as my development machine. I followed the instructions on the renode github page and installed renode and mono. However when i launch the renode console and then load stm32f4_discovery-kit.repl, i get the following error.

`[ERROR] FATAL UNHANDLED EXCEPTION: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.InvalidOperationException: Error while opening dynamic library: dlopen(/var/folders/pw/1s5vzcl575l6vgsm2nvjz59h0000gn/T/renode-19465/0-Antmicro.Renode.translate-arm-m-le.so, 0x0002): tried: '/var/folders/pw/1s5vzcl575l6vgsm2nvjz59h0000gn/T/renode-19465/0-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/var/folders/pw/1s5vzcl575l6vgsm2nvjz59h0000gn/T/renode-19465/0-Antmicro.Renode.translate-arm-m-le.so' (no such file), '/var/folders/pw/1s5vzcl575l6vgsm2nvjz59h0000gn/T/renode-19465/0-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/private/var/folders/pw/1s5vzcl575l6vgsm2nvjz59h0000gn/T/renode-19465/0-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/private/var/folders/pw/1s5vzcl575l6vgsm2nvjz59h0000gn/T/renode-19465/0-Antmicro.Renode.translate-arm-m-le.so' (no such file), '/private/var/folders/pw/1s5vzcl575l6vgsm2nvjz59h0000gn/T/renode-19465/0-Antmicro.Renode.translate-arm-m-le.so' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
  at Antmicro.Renode.Utilities.SharedLibraries.HandleError (System.String operation) [0x00031] in <a3a25a8c0fe34b6296404c54bf56a731>:0
  at Antmicro.Renode.Utilities.SharedLibraries.LoadLibrary (System.String path) [0x0000a] in <a3a25a8c0fe34b6296404c54bf56a731>:0
  at Antmicro.Renode.Utilities.Binding.NativeBinder..ctor (Antmicro.Renode.IEmulationElement classToBind, System.String libraryFile) [0x00025] in <a3a25a8c0fe34b6296404c54bf56a731>:0
  at Antmicro.Renode.Peripherals.CPU.TranslationCPU.Init () [0x000b0] in <ab4cdcddd78e4c6ca1c68b4e84296edb>:0
  at Antmicro.Renode.Peripherals.CPU.TranslationCPU..ctor (System.UInt32 id, System.String cpuType, Antmicro.Renode.Core.IMachine machine, ELFSharp.ELF.Endianess endianness, Antmicro.Renode.Peripherals.CPU.CpuBitness bitness) [0x00083] in <ab4cdcddd78e4c6ca1c68b4e84296edb>:0
  at Antmicro.Renode.Peripherals.CPU.Arm..ctor (System.String cpuType, Antmicro.Renode.Core.IMachine machine, System.UInt32 cpuId, ELFSharp.ELF.Endianess endianness, System.Nullable`1[T] numberOfMPURegions) [0x00060] in <95379f190e174f14a876f9cfc1dbbc95>:0
  at Antmicro.Renode.Peripherals.CPU.CortexM..ctor (System.String cpuType, Antmicro.Renode.Core.IMachine machine, Antmicro.Renode.Peripherals.IRQControllers.NVIC nvic, System.UInt32 id, ELFSharp.ELF.Endianess endianness, System.Nullable`1[T] fpuInterruptNumber, System.Nullable`1[T] numberOfMPURegions) [0x00007] in <b2f3c18913df418fa3cd31edc812b01e>:0
  at (wrapper managed-to-native) System.Reflection.RuntimeConstructorInfo.InternalInvoke(System.Reflection.RuntimeConstructorInfo,object,object[],System.Exception&)
  at System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) [0x00005] in <83a10ac9d03d4b5b9cab686735823828>:0`

### Expected behaviour

I should be able to load and launch the platform. 

### How to reproduce?

1. renode --console
2. mach create
3. machine LoadPlatformDescription @platforms/boards/stm32f4_discovery-kit.repl

### Environment

- OS : Mac OS Sonoma 14.1.1
- Renode version (with commit SHA) : Renode, version 1.14.0.6718 (512e96cf-202311210343)
- Mono version :  Mono JIT compiler version 6.12.0.182 (tarball Mon Jan 16 21:39:07 UTC 2023)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:
	SIGSEGV:       normal
	Notification:  kqueue
	Architecture:  arm64
	Disabled:      none
	Misc:          softdebug
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       preemptive
	GC:            sgen

princehektor, 2023-11-24:
Found the issue. It looks like i was launching the wrong version of the renode application. I initially followed this link(https://github.com/renode/renode) and set up the alias for renode as 

alias renode='mono /Applications/Renode.app/Contents/MacOS/bin/Renode.exe'

While renode did launch with this, i hit the error mentioned above when i tried to load the platform.

I changed my alias to `alias renode='sh /Applications/Renode.app/Contents/MacOS/macos_run.command'` and it works fine now.

princehektor, 2023-11-24:
Fixed the issue myself. Resolution mentioned in previous comment.

closed 2023-11-24 by princehektor

db-ae2397, 2024-05-22:
I had the same initial conditions as you had, but this solution didn't work for me.

dbarwacz, 2024-07-19:
I had a similar problem running renode-test (robot). I narrowed it down to how renode vs renode-test scripts are configured. Turns out [renode](https://github.com/renode/renode/blob/master/tools/packaging/macos/macos_run.command#L43) runs on `mono64` while [renode-test](https://github.com/renode/renode/blob/master/tests/robot_tests_provider.py#L369) runs on `mono` (hence binary arch compatibility errors). 
Changing second linked line to `command.insert(0, 'mono64')` fixed the problem for me.