renode/renode #556: Uart not showing up

MTsimon, 2023-11-24:
### Description

MCU: NRF52840
Operating system: zephyr-v3.4.0-77-g75293c8c79eb

I have a simple "hello world" application, with some driver initialization for a custom board.
Program .hex file loads fine on target, but when loading the same .hex file to renode, I dont see anything on the UART. 

main.c:
```cpp
int main(void)
{
	LOG_INF("Custom board");
	while (1) {
		LOG_INF("Hello");
		k_sleep(K_SECONDS(1));
	}
	return 0;
}
```

My .repl file looks like this:

```
mach add "custom-board"
machine LoadPlatformDescription @platforms/cpus/nrf52840.repl
showAnalyzer sysbus.uart1

macro reset
"""
    sysbus LoadHEX @app/build/zephyr/zephyr.hex
"""
runMacro $reset
```

What I see on renode debug console, is that the machine is constantly pausing/resuming with this output:

```

08:30:58.6239 [INFO] cpu: Setting initial values: PC = 0xC2E1, SP = 0x200286D8.
08:30:58.6269 [INFO] custom-board: Machine resumed.
08:30:58.6269 [WARNING] sysbus: [cpu: 0x299C0] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000.
08:30:58.6279 [WARNING] sysbus: [cpu: 0x2998C] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000. (4)
08:30:58.6279 [WARNING] sysbus: [cpu: 0x29ABC] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000.
08:30:58.6279 [WARNING] sysbus: Read from an unimplemented register UICR:NFCPINS (0x1000120C), returning a value from SVD: 0xFFFFFFFF.
08:30:58.6289 [WARNING] sysbus: Write of value 0x1 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
08:30:58.6289 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.6289 [WARNING] sysbus: Read from an unimplemented register UICR:NFCPINS (0x1000120C), returning a value from SVD: 0xFFFFFFFF.
08:30:58.6289 [WARNING] sysbus: Write of value 0xFFFFFFFE to an unimplemented register UICR:NFCPINS (0x1000120C) generated from SVD.
08:30:58.6289 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.6289 [WARNING] sysbus: Write of value 0x0 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
08:30:58.6289 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.6289 [INFO] custom-board: Machine paused.
08:30:58.7259 [INFO] cpu: Setting initial values: PC = 0xC2E1, SP = 0x200286D8.
08:30:58.7289 [INFO] custom-board: Machine resumed.
08:30:58.7289 [WARNING] sysbus: [cpu: 0x299C0] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000.
08:30:58.7309 [WARNING] sysbus: [cpu: 0x2998C] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000. (4)
08:30:58.7309 [WARNING] sysbus: [cpu: 0x29ABC] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000.
08:30:58.7309 [WARNING] sysbus: Read from an unimplemented register UICR:NFCPINS (0x1000120C), returning a value from SVD: 0xFFFFFFFF.
08:30:58.7319 [WARNING] sysbus: Write of value 0x1 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
08:30:58.7319 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.7319 [WARNING] sysbus: Read from an unimplemented register UICR:NFCPINS (0x1000120C), returning a value from SVD: 0xFFFFFFFF.
08:30:58.7319 [WARNING] sysbus: Write of value 0xFFFFFFFE to an unimplemented register UICR:NFCPINS (0x1000120C) generated from SVD.
08:30:58.7319 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.7319 [WARNING] sysbus: Write of value 0x0 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
08:30:58.7319 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.7329 [INFO] custom-board: Machine paused.
08:30:58.8230 [INFO] cpu: Setting initial values: PC = 0xC2E1, SP = 0x200286D8.
08:30:58.8259 [INFO] custom-board: Machine resumed.
08:30:58.8259 [WARNING] sysbus: [cpu: 0x299C0] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000.
08:30:58.8279 [WARNING] sysbus: [cpu: 0x2998C] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000. (4)
08:30:58.8279 [WARNING] sysbus: [cpu: 0x29ABC] (tag: 'FICR') ReadDoubleWord from non existing peripheral at 0x10000130, returning 0x00000000.
08:30:58.8279 [WARNING] sysbus: Read from an unimplemented register UICR:NFCPINS (0x1000120C), returning a value from SVD: 0xFFFFFFFF.
08:30:58.8279 [WARNING] sysbus: Write of value 0x1 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
08:30:58.8290 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.8290 [WARNING] sysbus: Read from an unimplemented register UICR:NFCPINS (0x1000120C), returning a value from SVD: 0xFFFFFFFF.
08:30:58.8290 [WARNING] sysbus: Write of value 0xFFFFFFFE to an unimplemented register UICR:NFCPINS (0x1000120C) generated from SVD.
08:30:58.8290 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.8290 [WARNING] sysbus: Write of value 0x0 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
08:30:58.8290 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
08:30:58.8290 [INFO] custom-board: Machine paused.
```

Loading an example .elf file works fine. 

Can someone point me, if I am doing something wrong, or if renode is not supported for my specific board/drivers?

I see a alot of warning messages, but no errors?

PiotrZierhoffer, 2023-11-30:
Hi @MTsimon ,

Thanks for reporting the issue. I believe it boils down to the fact that you never reached the `main` function.

This can be easily verified. Can you add `cpu LogFunctionNames true true` to your script and post the log?

BettaSharma, 2024-10-17:
Hi @PiotrZierhoffer,

Since the problem still appears open, and I seem to be seeing same problem (UART0 not working?), I have attached both the monitor commands and log.

Renode, version 1.15.3.22387 (03756cb5-202409171221)
(monitor) mach create "my-machine"
(my-machine) machine LoadPlatformDescription @platforms/cpus/nrf52840.repl
(my-machine) peripherals
Available peripherals:
  sysbus (SystemBus)
  │   
  ├── clock (NRF52840_CLOCK)
  │       <0x40000000, 0x40000FFF>
  │       
  ├── cpu (CortexM)
  │       Slot: 0
  │       
  ├── ecb (NRF52840_ECB)
  │       <0x4000E000, 0x4000EFFF>
  │       
  ├── egu0 (NRF52840_EGU)
  │       <0x40014000, 0x40014FFF>
  │       
  ├── egu1 (NRF52840_EGU)
  │       <0x40015000, 0x40015FFF>
  │       
  ├── egu2 (NRF52840_EGU)
  │       <0x40016000, 0x40016FFF>
  │       
  ├── egu3 (NRF52840_EGU)
  │       <0x40017000, 0x40017FFF>
  │       
  ├── egu4 (NRF52840_EGU)
  │       <0x40018000, 0x40018FFF>
  │       
  ├── egu5 (NRF52840_EGU)
  │       <0x40019000, 0x40019FFF>
  │       
  ├── flash (MappedMemory)
  │       <0x00000000, 0x000FFFFF>
  │       
  ├── gpio0 (NRF52840_GPIO)
  │       <0x50000500, 0x500007FF>
  │       
  ├── gpio1 (NRF52840_GPIO)
  │       <0x50000800, 0x50000AFF>
  │       
  ├── gpiote (NRF52840_GPIOTasksEvents)
  │       <0x40006000, 0x40006FFF>
  │       
  ├── i2s (NRF52840_I2S)
  │       <0x40025000, 0x40025FFF>
    │       
  ├── nvic (NVIC)
  │       <0xE000E000, 0xE000EFFF>
  │       
  ├── pdm (NRF52840_PDM)
  │       <0x4001D000, 0x4001DFFF>
  │       
  ├── ppi (NRF52840_PPI)
  │       <0x4001F000, 0x4001FFFF>
  │       
  ├── radio (NRF52840_Radio)
  │       <0x40001000, 0x40001FFF>
  │       
  ├── ram (MappedMemory)
  │       <0x00800000, 0x0083FFFF>
  │       <0x20000000, 0x2003FFFF>
  │       
  ├── rng (NRF52840_RNG)
[... 708 lines cut ...]
            printf(" \r\nExit!\r\n");

            while (true)
            {
                // Do nothing.
            }
        }
    }
#else

    // This part of the example is just for testing the loopback .
    while (true)
    {
        uart_loopback_test();
    }
#endif
}

/** @} */
----------------------------------------------------------------------------------------------------------------------------

This program builds and prints 
UART example started.

On a Termite 3.4 Virtual COM Port terminal program when run on a BMD-340 board.