renode/renode #559: Local elf 'file does not exist' when running on macOS

dbarwacz, 2023-12-06:
Hi! 
I'm running `renode-test renode/renode.robot` from within a repository with a simple test case. 

When I try to run 
`sysbus LoadELF @0-target.elf` 
I encounter an issue `KeywordException: Could not execute command 'sysbus LoadELF @0-target.elf': There was an error executing command 'sysbus LoadELF @0-target.elf'File does not exist: 0-target.elf` 

The very same command works as expected when run from an interactive shell. I confirmed that the working dir of `renode-test` is  `Renode.app/Contents/MacOS/bin` (when I add the file there, it works), rather than pwd where I run `renode-test` (which is the case for an interactive shell).

I think there's an issue with my local installation or how renode is hooked up. I followed the instructions of README (aliasing renode and renode-test contained in Renode.app package). 

Context: 
* the platform you are using: MacOS
* the Renode version you are using: 1.14.0

Should I change my installation or can I pass a parameter to specify where renode-test should look up my resources? 
Thanks! 

Btw I really love the idea and execution of this OSS, you rock!

PiotrZierhoffer, 2023-12-14:
Hi @dbarwacz ,

Thank you for your kind words.

I think the best way to address this would be to follow this pattern: https://github.com/renode/renode/blob/master/tests/unit-tests/log-tests.robot#L77

You need to use `${CURDIR}` variable in the robot script to use a file from the same directory. The `${/}` part is important if you run also on Windows.

dbarwacz, 2023-12-15:
That works for me, I must have missed that when I was briefly looking at examples. Thanks!

closed 2023-12-15 by dbarwacz