renode/renode #562: How to use/enable STM32 timer channels

waltergallegog, 2023-12-17:
I would like to use the channels features of STM32 timers, but I'm unsure if this is already supported, or how to use them.

For example, Timer10 of the STM32F4 provides channels with different configurations, mainly:
 - Output compare, to generate PWMs.
 - Input capture, to measure events.

In `STM32_Timer.cs` I see that these features seem to already be in place, however I have tried them but they did not work.

Both of these features require the use of a GPIO, in output or input mode, respectively. The GPIO to use is fixed and different for each timer, as can be seen when configuring the firmware, for example from CubeMX. However, from the renode platform description examples of the STM32 family, I can not find a platform that actually connects any of the timers to the required GPIO. 

- Is the channels feature supported?
- How can I enable/use it?
- How can I indicate the GPIO to use for each Timer?

Thanks.

htamas109, 2025-02-06:
Hello Walter,

Were you able to solve your problem?
I'm seemingly facing the same issue, and I would really appreciate it if you could share your experience.

Thank you very much in advance!

Best regards,
Tamás

waltergallegog, 2025-02-06:
Hi Tamás
Unfortunately not, so for now I'm using the timers just in counting mode.
BR
Walter

htamas109, 2025-02-06:
Hi Walter,

Thank you for your answer. That's unfortunate.
I will leave a comment, if I can figure out any solution.

Best regards,
Tamás