renode/renode #563: UI won't start on MacOS

mick-h, 2023-12-20:
I've installed Renode 1.14 on an Intel Mac running MacOS 13.1.  When I run it I get the following output:

```shell
Last login: Wed Dec 20 13:53:50 on ttys000
###@#### ~ % /Applications/Renode.app/Contents/MacOS/macos_run.command ; exit;
14:08:10.9327 [WARNING] Couldn't start UI - falling back to console mode
14:08:11.4492 [INFO] Loaded monitor commands from: /Applications/Renode.app/Contents/MacOS/scripts/monitor.py
Renode, version 1.14.0.28928 (81da46c1-202308081604)

(monitor)
``` 

Mono is version 6.12.0.182 installed via brew.  Any ideas on how to get the UI running?

xobs, 2024-01-18:
You'll notice that it also crashes if you actually try to start any emulation.

Renode's emulation core seems to not work at all on ARM64 targets. The .NET stuff works fine, but the compiled code just doesn't know how to run on anything other than x64. 

When you first ran Renode it should have taken you to a link to install the official Microsoft .NET runtime. If you install this, Renode will run under Rosetta2 and will work normally.

mick-h, 2024-01-19:
It's an Intel Mac (circa 2019).  I'm not sure if I had to install .Net as I have Mono installed via brew (I don't seem to have the Microsoft .Net installed AFAICT.  Simulations do run (with a truckload of warnings about unhandled writes to mmc and sysbus reads from non-existent peripherals, and it hangs for an eternity after starting mdev:

`15:17:52.2470 [INFO] mmuart1: [host:    0.12ks (+0.3s)|virt: 6.61s (+23.2ms)] Starting mdev... OK`

when running the @scripts/single-node/icicle-kit.resc.  It has eventually got to a linux prompt a couple of times, but quite a few times I have waited many many minutes at that prompt and then eventually killed it.  Not sure if that is related to the same cause as the UI issue, but more to show that the emulations do run, if not well.

Should I install the official MS .Net runtime over Mono and try that?

PiotrZierhoffer, 2024-03-21:
@mick-h the package you have been using is quite old. The new  Renode version, e.g. 1.15.0 or any new taken from https://builds.renode.io/, should work fine.

Re icicle - this is an unrelated issue. We do observe an instability there and it is being investigated separately.

Regarding runtime - on macOS right now we're using Mono only, so I recommend sticking to it. We are planning to release dotnet support soon as well.

mick-h, 2024-03-22:
I must be doing something special...

```
Last login: Sat Mar  9 23:23:52 on ttys000
XXX@XXXX ~ % /Applications/Renode.app/Contents/MacOS/macos_run.command ; exit;
02:02:44.5662 [WARNING] Couldn't start UI - falling back to console mode
02:02:45.2588 [INFO] Loaded monitor commands from: /Applications/Renode.app/Contents/MacOS/scripts/monitor.py
Renode, version 1.15.0.6706 (639524d4-202403210343)

(monitor)
```

Joebeazelman, 2025-08-07:
I'm getting the same issue.

mjr-tamu, 2026-04-14:
@lmapii, you're spot on. Thank you for posting this.