renode/renode #564: Loading PythonPeripheral script from an external .py file

slosar, 2023-12-20:
### Description

PythonPeripheral is exactly what I need for my work, but the inline nature of the script max it difficult to go beyond the simplest script. At the minimum support reading a multiline string from file would help, but a more custom syntax would be even better (I couldn't find this option despite reading the doc carefully)

At the moment I am solving this with the following work-around:

```
dformatter: Python.PythonPeripheral @ sysbus 0x80100000
    size: 0x0100000
    initable: true
    script: '''
if request.isInit:
    import sys, os
    sys.path.append('/home/anze/work/lusee/LuSEE_MiV/renode')
    from DataFormatter import DataFormatter
    dformatter = DataFormatter()
else:
    dformatter.access(self, request, machine )
'''
```
This works ok, but it is totally non-portable since I haven't worked out how to specify the python path relative to the place of repl file. pwd = /opt/renode when this scripts gets executed.

### Usage example

Ideally I would like something like:

```
dformatter: Python.PythonPeripheral @ sysbus 0x80100000
    size: 0x0100000
    module: py/DataFormatter.py
    object: DataFormatter
```
This would initialize DataFormatter object and call some method in it ever time the peripheral is requested.

Somewhat less desirable:

```
dformatter: Python.PythonPeripheral @ sysbus 0x80100000
    size: 0x0100000
    initable: true
    script:  @'DataFormatter.py'
```
(for some syntax to denote reading script from file)

### Do you plan to address this issue and file a PR?

I might if it becomes too cumbersome, but currently too new to renode.

mateusz-holenko, 2024-01-12:
Hi @slosar, thanks for asking the question.

In fact it is already possible to provide a script to `PythonPeripheral` from external file (like you suggested in the second option). 
In order to do that you need to use `filename` parameter instead of `script`, though.
Please see an example in: https://github.com/renode/renode/blob/0e6a89cc35669a2f671ef7d89aefe23a167ef3ce/platforms/cpus/zynqmp.repl#L43

You are right we are missing this piece of information in our docs, will fix it shortly!

slosar, 2024-01-12:
Great, thanks. It would be good if this was documented somewhere, I scanned all over the documentation and haven't seen this anywhere.

mateusz-holenko, 2024-01-18:
We've updated the documentation to include the `filename` parameter in the list.

closed 2024-01-18 by mateusz-holenko