renode/renode #571: Rust binding for Renode

asmitaj08, 2023-12-31:
Hi,
Is it possible to create Rust binding for Renode?
If yes, any suggestiins regarding how to proceed ?

Thank you!!

WardBenjamin, 2023-12-31:
Check out this project: https://github.com/jonlamb-gh/renode-run

ratmice, 2025-05-02:
In the blog post linked above, the example links to this branch/fork[^1] of renode, but that includes a `WebAssembly.dll`,
So it doesn't seem like something that has worked on any of the linux portable releases.

[^1]: https://github.com/renode/renode/compare/master...26999-rust_uart

I've been interested in how/whether it is possible to use the rust `defmt` crate which is often used to format data sent through the `rtt` ring buffers, so I'd be curious to know if there were any thoughts on what needs to be done to get this into the main branches, and how or what we need to do to get the wasm dll on other platforms?